flyte/mqtt-io #1: AttributeError: 'module' object has no attribute 'REQUIREMENTS'

Genestealer, 2017-02-26:
Error when trying to run for the 1st time:

```
sudo python -m pi_mqtt_gpio.server /home/pi/code/config.yml
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 96, in <module>
    install_missing_requirements(gpio_module)
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 29, in install_missing_requirements
    reqs = getattr(module, "REQUIREMENTS")
AttributeError: 'module' object has no attribute 'REQUIREMENTS'
```

RPi.GPIO is already installed:
```
pip install RPi.GPIO
Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages
```

Config file:
```
mqtt:
  host: localhost
  port: 1883
  user: ""
  password: ""
  topic_prefix: pimqttgpio/mydevice

gpio_modules:
  - name: raspberrypi
    module: raspberrypi
  - name: dev
    module: stdio

digital_inputs:
  - name: button
    module: raspberrypi
    pin: 21
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes

digital_outputs:
  - name: test
    module: dev
    pin: 1
    on_payload: "ON"
    off_payload: "OFF"
```

Any ideas?

Genestealer, 2017-02-26:
Same error when trying the other example:

Config file:
```
mqtt:
  host: test.mosquitto.org
  port: 1883
  user: ""
  password: ""
  topic_prefix: home

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_inputs:
  - name: doorbell
    module: raspberrypi
    pin: 22
    on_payload: "ON"
    off_payload: "OFF"
    pullup: yes
    pulldown: no
```

Output:
```
pi@raspberrypi:~ $ python -m pi_mqtt_gpio.server /home/pi/code/config.yml
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 111, in <module>
    for output_config in config["digital_outputs"]:
KeyError: 'digital_outputs'
```

flyte, 2017-02-26:
Looks like you found a bug (or two)!
- I didn't make REQUIREMENTS optional for the modules, and then didn't put it in the stdio module.
- The code makes an assumption that both the `digital_inputs` and `digital_outputs` config sections exist.

These should now be fixed, but I don't have a Pi handy for testing it out.

Please let me know of any further issues :)

flyte, 2017-02-26:
Closing the bug for now, as I've tested it superficially without the `raspberrypi` module. I may endeavour to tidy this project up a bit if someone else is actually using it :)

flyte, 2017-02-26:
I also forgot to mention that I've just pushed a new version (0.0.3) including this fix. You should now be able to do a `pip install --upgrade pi-mqtt-gpio`

closed 2017-02-26 by flyte