flyte/mqtt-io #4: PWM

dr-apple, 2017-05-21:
Hello,

it works all very good, thanks for your script...

Could you still install PWM for all GPIO?

flyte, 2017-06-13:
I certainly can. I'll have a think about how to implement it and get it done when I've got a little time to spare.

dr-apple, 2017-06-15:
Perfectly thank you, you can also times times whether you can speed up the input of an input? It takes about 1 second until the change is recognized :-( This is too slow at a pushbutton ....

flyte, 2017-06-15:
Yep, it's on the list - see #5 :)

Gr4ffy, 2021-05-12:
Hi,
Great job @flyte 
Is PWM implemented already?

flyte, 2021-05-12:
Hey @Gr4ffy thanks!

It's currently not implemented I'm afraid. There's no technical reason why it couldn't be, just not enough time in the day 😊

Gr4ffy, 2021-05-12:
Got it, thanks for the feedback.

bosu1787, 2022-04-23:
no pwm yet?

kc2zgu, 2024-05-17:
I think the `/sys/class/pwm` interface makes the most sense (instead of trying to write several drivers for specific hardware when there is already a common API in the kernel), and to be a dedicated module that doesn't try to figure out things like mappings that may not even exist between regular GPIO pins and PWM pins. Just rely on the user to determine which PWM chip and line they want to use and have the driver loaded. I've used this interface from C before, all you have to do it write the pin number to the `export` file (like the old sysfs gpio), set the frequency/period with `period`, write 1 to `enable`, and write a number to `duty_cycle` to set the pulse length.