flyte/mqtt-io #6: Problem with PCF8574

dr-apple, 2017-07-01:
Here my problem:

python -m pi_mqtt_gpio.server config.yml
Downloading/unpacking pcf8574
  Downloading pcf8574-0.0.5.tar.gz
  Running setup.py (path:/tmp/pip-build-0xzCvc/pcf8574/setup.py) egg_info for package pcf8574
    your setuptools is too old (<12)
    setuptools_scm functionality is degraded
    zip_safe flag not set; analyzing archive contents...
    
    Installed /tmp/pip-build-0xzCvc/pcf8574/pytest_runner-2.11.1-py2.7.egg
    
Downloading/unpacking smbus-cffi (from pcf8574)
  Downloading smbus-cffi-0.5.1.tar.gz
  Running setup.py (path:/tmp/pip-build-0xzCvc/smbus-cffi/setup.py) egg_info for package smbus-cffi
    
    Installed /tmp/pip-build-0xzCvc/smbus-cffi/cffi-1.10.0-py2.7-linux-armv7l.egg
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip-build-0xzCvc/smbus-cffi/setup.py", line 48, in <module>
        'Topic :: System :: Hardware',
      File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 266, in __init__
        _Distribution.__init__(self,attrs)
      File "/usr/lib/python2.7/distutils/dist.py", line 287, in __init__
        self.finalize_options()
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 301, in finalize_options
        ep.load()(self, ep.name, value)
      File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
        ['__name__'])
    ImportError: No module named setuptools_ext
    Complete output from command python setup.py egg_info:
    

Installed /tmp/pip-build-0xzCvc/smbus-cffi/cffi-1.10.0-py2.7-linux-armv7l.egg

Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip-build-0xzCvc/smbus-cffi/setup.py", line 48, in <module>

    'Topic :: System :: Hardware',

  File "/usr/lib/python2.7/distutils/core.py", line 111, in setup

    _setup_distribution = dist = klass(attrs)

  File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 266, in __init__

    _Distribution.__init__(self,attrs)

  File "/usr/lib/python2.7/distutils/dist.py", line 287, in __init__

    self.finalize_options()

  File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 301, in finalize_options

    ep.load()(self, ep.name, value)

  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load

    ['__name__'])

ImportError: No module named setuptools_ext

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip-build-0xzCvc/smbus-cffi
Storing debug log for failure in /home/pi/.pip/pip.log
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 99, in <module>
    GPIOS[gpio_config["name"]] = gpio_module.GPIO(gpio_config)
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/modules/pcf8574.py", line 23, in __init__
    from pcf8574 import PCF8574
ImportError: No module named pcf8574

flyte, 2017-07-03:
Could you try:
```
pip install pcf8574
```
and running it again?

flyte, 2017-07-28:
Hey @dr-apple I'm closing this for now as I'm not sure there's an issue here with this project. Please comment back if the problem persists and it looks like I'm wrong.

closed 2017-07-28 by flyte

aerodolphin, 2020-10-16:
I installed the next modules:

```
pip install pcf8574
pip install pcf8575
```
But I'm getting always the same error in the log:

`ImportError: No module named pcf8574`

or

`ImportError: No module named pcf8575`

What could be the problem?

aerodolphin, 2020-10-16:
When I try to remove the modules with command:

`sudo apt-get --purge remove pcf8574`

I got this message:

`E: Unable to locate package pcf8574`

It means that the package was not installed?