flyte/mqtt-io #7: Question: Multiple output statements

Paul-dH, 2017-07-02:
Hi,

Thanks for writing the module! This makes life a lot easier :) 

At the moment I'm writing a small how-to on fully deploying a Pi with 4 realys and connecting them to Home Assistant (for automating a ceiling fan).

My question is: Is it possible to create a config file containing multiple output statements? Like speed-1, speed-2, speed-3, lights. Or should they get their own topics, it's all for the same device.

At the moment my config looks like:

`mqtt:
  host: 192.xxx.xxx.xxx
  port: 1883
  user: "redacted"
  password: "redacted"
  topic_prefix: bedroom/ceiling-fan

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: speed-1
    module: raspberrypi
    pin: 17
    on_payload: "ON"
    off_payload: "OFF"

digital_outputs:
  - name: speed-2
    module: raspberrypi
    pin: 18
    on_payload: "ON"
    off_payload: "OFF"

digital_outputs:
  - name: lights
    module: raspberrypi
    pin: 23
    on_payload: "ON"
    off_payload: "OFF"

One more ...`

Thanks for your input!

flyte, 2017-07-03:
Hi, you're very welcome and I'm glad it's useful for you!

I'm not 100% sure I understand your setup, but I'll try answering anyway. To configure it for multiple digital outputs, you can use one `digital_output` section and list each output under it, something like this:

```yaml
mqtt:
  host: 192.xxx.xxx.xxx
  port: 1883
  user: "redacted"
  password: "redacted"
  topic_prefix: bedroom/ceiling-fan

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: speed-1
    module: raspberrypi
    pin: 17
    on_payload: "ON"
    off_payload: "OFF"

  - name: speed-2
    module: raspberrypi
    pin: 18
    on_payload: "ON"
    off_payload: "OFF"

  - name: lights
    module: raspberrypi
    pin: 23
    on_payload: "ON"
    off_payload: "OFF"
```

Then a publish to `bedroom/ceiling-fan/speed-1` with payload of `ON` should turn on pin 17, and a publish to `bedroom/ceiling-fan/speed-2` with payload of `ON` should turn on pin 18 etc.

Does that answer your question?

I'd be interested to see the how-to once it's done - stick a link in this issue and I'll check it out :)

Paul-dH, 2017-07-03:
Thanks for the answer! The topics posted on the commandline after the server start seems right. I'll keep you posted about the how-to, still figuring out how to get mqtt and Home Assistant to play nicely together :grin:

closed 2017-07-03 by Paul-dH

flyte, 2017-07-03:
Excellent. Good luck!