flyte/mqtt-io #8: Feature request

Paul-dH, 2017-07-03:
Hi @flyte 

Is it possible to add the options to provide a protocol and a websocket port? Home Assistant requires these to function :simple_smile:

Below is a example of the defaults that should be entered in a config [Link](https://home-assistant.io/docs/mqtt/broker/)

```
Host: localhost
Port: 1883
Protocol: 3.1.1
User: homeassistant
Password: Your API password
Websocket-port: 8080
```

Paul-dH, 2017-07-03:
Sorry, this issue can be closed. I had to follow numerios tutorials to get a working situation, but this had nothing to do with the code.

I'll do have a cool new feature request btw (for relays to require pins to be in off state before putting the called relay in on state)

Regards, Paul

PS: Sorry for the spam :sweat_smile:

closed 2017-07-03 by Paul-dH

flyte, 2017-07-03:
No probs. I did have a quick look at this and I wasn't sure what you were getting at :)