flyte/mqtt-io #9: Feature request:

Paul-dH, 2017-07-04:
Hi @flyte,

Is it possible to add something like this:

```
digital_outputs:
  - name: speed-1
    module: raspberrypi
    require_pin_off: 18,22
    pin: 17
    on_payload: "ON"
    off_payload: "OFF"
  - name: speed-2
    module: raspberrypi
    require_pin_off: 17,22
    pin: 18
    on_payload: "ON"
    off_payload: "OFF"
```

The reason for this is that I'm using 3 relays for 3 speeds of a ceiling fan. When speed-1 is on and speed-2 is started the pin for speed-1(17) should first be disabled before the pin of speed-2(18) is enabled. Else there are two relays in ON state and I guess the fan won't like this.

Or is there a better way to do this?

Thanks again for the effort!

flyte, 2017-07-04:
I expect implementing an `exclusive_on` and `exclusive_off` might do the trick. Each list can contain a list of output names which must be switched on/off before the pin is changed. 

For example:
```yaml
digital_outputs:
  - name: speed-1
    module: raspberrypi
    pin: 17
    on_payload: "ON"
    off_payload: "OFF"
    exclusive_on:
      - speed-2

  - name: speed-2
    module: raspberrypi
    pin: 18
    on_payload: "ON"
    off_payload: "OFF"
    exclusive_on:
      - speed-1
```

I'd need to think about how to avoid conflicting exclusivity when using `exclusive_on` AND `exclusive_off`, but this approach may work.

Paul-dH, 2017-07-04:
Yes I think we're on the same page here. Maybe require_on or off gives the option a better understanding, but it's just the name. The option to give a requirement of a section being in off state is indeed more clear than just the pin. 
Maybe I can help somewhere needed? 
Regards, Paul