flyte/mqtt-io #10: Toggle output

meifi, 2017-07-05:
Hello, 

i use some pcf8574 from horter (http://www.horter.de/i2c/i2c-output-modul/i2c-output-modul-Beschreibung.html) to controll surge relays. 

It is possible to add a function to the options with witch we can activate a automatic reset of the port? When i set port 1 to ON and the value "TOGGLE" is activated in the config the port reset it selfs after x MS to OFF autmaticly. 

thanks

flyte, 2017-07-05:
Just so I'm clear, do you mean to turn the output on for x milliseconds and then return to its previous state? If so, I think the implementation might be best done with a publish to `.../set_ms` or similar, instead of `.../set`. Would that suit you?

meifi, 2017-07-06:
hi flyte, 

yes it woud be perfect if we can publish to /set to set the output ON oder OFF until the next publish and when publish to /toggle the output set for x MS to ON and reset to the previous state . so we can set the MS for the toggle in the config file. It is also posible to publish the MS value to /toggle_ms as payload. 

what do you think is easier to implementate?

flyte, 2017-07-06:
I think perhaps `/set_on_ms` and `/set_off_ms` might be simpler for digital outputs. This way there's only a single value required in the payload.

What do you think about which value they should end up as? If the output is ON already, then we `/set_on_ms` with `1000` as the payload, should the output turn OFF after 1 second, or stay ON, as it was already?

meifi, 2017-07-06:
I think it must be the last value. the reason is, that some modules has a invertet logic. 

If the output is ON already, then we /set_on_ms with 1000 as payload, the output change for 1000ms to OFF and return to the last value - ON.

flyte, 2017-07-17:
Reopening as a reminder to update the readme with instructions on using the new feature.

flyte, 2017-07-28:
@meifi I've added an `inverted` config value for digital outputs which avoids the problem with explicit 'on' and 'off' topic suffixes. (See #18) and changed the behaviour of the `/set_on_ms` and `/set_off_ms` topics to always end up in the opposite state.

So for example, a light is ON, you publish to `/set_on_ms` and after the timeout, the light turns OFF.

If the light is OFF to begin with, the light will turn ON immediately, then turn OFF after the timeout.

Make sense?

commit ae4d5603 (2017-07-17): Add set_on_ms and set_off_ms topics. Closes #10
commit a7fe9c3f (2017-07-17): Add set_on_ms and set_off_ms topics. Closes #10
closed 2017-07-28 by flyte