flyte/mqtt-io #12: GPIO init at boottime

Paul-dH, 2017-07-07:
When the RPI gets a reboot the relay swtiches ON for 0.700ms. I think this has something to do with the server being started.

I've searched on Google what this might be, there are some pins (1-8) that are set to HIGH at boot time, and the rest are set to LOW, documentation is found here (See page 102):
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf

I've changed the pins of the relay to GPIO 6, 12, 26 and 20 and when I remove the mqtt-server from crontab they stay OFF. But when I uncomment the rule again and the server starts the relays are pulled.

I've also tried to include a script at boot but that didn't work:
```
#!/bin/sh

echo "2" > /sys/class/gpio/export
gpio -g write 2 1
gpio -g export 2 out
```
But then again, I might use the script above in the wrong way.

flyte, 2017-07-17:
So when you disable this project from launching at boot, the outputs do not switch on? Do they switch on briefly when you run this software manually afterwards?

Paul-dH, 2017-07-22:
I've done some testing and when I disabled the crontab entry that starts the server the relays stay off. When I start the server manualy you can see that it subscribes to the topics again and all the relays are switched on. When the MQTT server (Home Assistant) is shutdown and unreachable the server on the Pi generates a error (see below) and all relays are switched on again.

`pi@fan-1:~ $ sudo /usr/bin/python -m pi_mqtt_gpio.server /var/scripts/mqtt-config.yml
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 118, in <module>
    client.connect(config["mqtt"]["host"], config["mqtt"]["port"], 60)
  File "/usr/local/lib/python2.7/dist-packages/paho_mqtt-1.3.0-py2.7.egg/paho/mqtt/client.py", line 760, in connect
    return self.reconnect()
  File "/usr/local/lib/python2.7/dist-packages/paho_mqtt-1.3.0-py2.7.egg/paho/mqtt/client.py", line 887, in reconnect
    sock = socket.create_connection((self._host, self._port), source_address=(self._bind_address, 0))
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
socket.error: [Errno 113] No route to host`

I've also had an error in my setup that maybe relates to this behavior, my config of the server is like:

`mqtt:
  host: redacted
  port: 1883
  user: "redacted"
  password: "redacted"
  topic_prefix: homeassistant/slaapkamer/ventilator

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: speed-1
    module: raspberrypi
    pin: 6
    on_payload: "ON"
    off_payload: "OFF"

  - name: speed-2
    module: raspberrypi
    pin: 12
    on_payload: "ON"
    off_payload: "OFF"

  - name: speed-3
    module: raspberrypi
    pin: 26
    on_payload: "ON"
    off_payload: "OFF"

  - name: lights
    module: raspberrypi
    pin: 20
    on_payload: "ON"
    off_payload: "OFF"`

The config file should be good, and the configuration of Home Assistant matches these values. But the ON / OFF payloads are switched the other way arround. In Home Assistant the switch is ON, and the trace of the MQTT log als gives the ON state trough to the RPi, but the relay is switched OFF. So in order to switch OFF all relays, the swithes in Home Assistant must be put in ON state...

Could this be related, or did I made a f* up :)

flyte, 2017-07-26:
As far as the inversion of the ON/OFF values goes, this is probably just due to your relays being connected up as 'active low'. You could change the `on_payload` to `"OFF"` and `off_payload` to `"ON"` in order to fix this. That was my intended way of handling inverted logic like this.

It sounds like you may have retained messages on the MQTT server which switch the outputs as soon as this program is started up. I'm still not sure why you'd get a very brief on/off pulse though. Does it still happen when you invert the logic, as I've written above?

flyte, 2017-07-29:
@Paul-dH by the way, I've just implemented an `inverted` configuration value for digital outputs. Please see the updated readme, which includes a configuration example using `inverted`.

As far as this ticket goes, I'll eventually close it if I don't hear back from you, but please feel free to reply even if it's closed and I may reopen it for you.

Paul-dH, 2017-07-31:
Hi @flyte,

Sorry for the late response, I was a bit bussy last week. Thanks for the update, this looks like a good solution and I'll test this today. I'll report back when I've tested everything a couple of times.

Paul-dH, 2017-07-31:
Hi @flynth,

I've done some testing and wrote down everything below:

- Disabled start command of the server in crontab (relays stayed off)
- Enabled start command of the server in crontab: relays switched on after a while (switches were in OFF state in Home Assistant)
**Crontab entry:**
`*/2 * * * * /usr/bin/python -m pi_mqtt_gpio.server /var/scripts/mqtt-config.yml > /var/scripts/mqtt-server.log 2>&1`

- Changed payload on to off and off to on (this caused the relays to stay off but they turned on for a couple of milliseconds at the mqtt-server startup)
- Changed all the pins to the old locations and changed payload on to off and off to on. This was to test weather the other "old" pins had a different boot state. But the same result, at boot time the relays turned on for a couple of milliseconds:
   pins old --> new location
```
    20 --> 23
    26 --> 22
    12 --> 18
    06 --> 17
```

- Home Assistant reports the right status of the relays (off, and the relay is off)
- Added the `inverted: yes` option to the config and reboot, the relays stayed off but they turned on for a couple of milliseconds at the mqtt-server startup.

To add a note, I'm using relay (http://modtronix.com/mod-rly2-5v.html) and a Raspberry Pi Zero W

It really seems that at boot time the server gives the pins a reset or something else that causes them to turn on for a couple of miliseconds.

I'm also thinking of changing the wirinig of the relays, I'm also reading that adding a transistor is possible to create a hardware failsafe for the relays. I only need to find some docs on how to do this :P

flyte, 2017-07-31:
The first thing I'd like to address is that running the program from a crontab is definitely wrong. You're essentially running a new copy of the program (as well as the existing one(s)) every two minutes. You only want to run one instance of the program. You may want to look into [supervisor](http://supervisord.org/) for this (`apt-get install supervisor`).

You say that "after a while" the relays switched on. This may be consistent with when a second copy of the program starts running after two minutes.

Could you try removing the crontab line (or disabling it), then run the program just once on the command line with `python -m pi_mqtt_gpio.server /var/scripts/mqtt-config.yml` and seeing if the relays still do stuff they're not supposed to?

Paul-dH, 2017-07-31:
I've disabled cron and installed supervisor. So after a reboot nothing is started yet, supervisor config is still empty. When I start the program the relays still jump on for a milliseccond. The output is below:

```
Subscribed to topic: 'homeassistant/slaapkamer/ventilator/output/speed-1/set'
Subscribed to topic: 'homeassistant/slaapkamer/ventilator/output/speed-2/set'
Subscribed to topic: 'homeassistant/slaapkamer/ventilator/output/speed-3/set'
Subscribed to topic: 'homeassistant/slaapkamer/ventilator/output/lights/set'
Got message on topic 'homeassistant/slaapkamer/ventilator/output/speed-1/set': 'OFF'
Set output 'speed-1' to True
Got message on topic 'homeassistant/slaapkamer/ventilator/output/speed-2/set': 'OFF'
Set output 'speed-2' to True
Got message on topic 'homeassistant/slaapkamer/ventilator/output/speed-3/set': 'OFF'
Set output 'speed-3' to True
Got message on topic 'homeassistant/slaapkamer/ventilator/output/lights/set': 'OFF'
Set output 'lights' to True
```

My config is:
```
mqtt:
  host: 10.x.x.x
  port: 1883
  user: "redacted"
  password: "redacted"
  topic_prefix: homeassistant/bedroom/ventilator

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: speed-1
    module: raspberrypi
    pin: 17
    inverted: yes
    on_payload: "OFF"
    off_payload: "ON"

  - name: speed-2
    module: raspberrypi
    pin: 18
    inverted: yes
    on_payload: "OFF"
    off_payload: "ON"

  - name: speed-3
    module: raspberrypi
    pin: 22
    inverted: yes
    on_payload: "OFF"
    off_payload: "ON"

  - name: lights
    module: raspberrypi
    pin: 23
    inverted: yes
    on_payload: "OFF"
    off_payload: "ON"
```

Paul-dH, 2017-07-31:
The problem still exists, I've added the `pullup: yes` option and also tried it without the `inverted: yes` parameter.

Below is the output when the server is started:
```
Subscribed to topic: 'homeassistant/slaapkamer/ventilator/output/speed-1/set'
Subscribed to topic: 'homeassistant/slaapkamer/ventilator/output/speed-2/set'
Subscribed to topic: 'homeassistant/slaapkamer/ventilator/output/speed-3/set'
Subscribed to topic: 'homeassistant/slaapkamer/ventilator/output/lights/set'
Got message on topic 'homeassistant/slaapkamer/ventilator/output/speed-1/set': 'OFF'
Set output 'speed-1' to True
Got message on topic 'homeassistant/slaapkamer/ventilator/output/speed-2/set': 'OFF'
Set output 'speed-2' to True
Got message on topic 'homeassistant/slaapkamer/ventilator/output/speed-3/set': 'OFF'
Set output 'speed-3' to True
Got message on topic 'homeassistant/slaapkamer/ventilator/output/lights/set': 'OFF'
Set output 'lights' to True
```

Maybe the relays are the problem, I've also read that the power source can be wired seperatly from the RPi.

I really appreciate the time you put in my debugging :), thanks!

flyte, 2017-11-07:
Hey @Paul-dH I've just released a new version in response to #23 which may hopefully fix these issues for you. You can now set an 'initial' state for each output pin when the software starts. I'll update the readme shortly for config examples.

I'm going to close this for now, as I'm optimistic, but please comment if you're still having problems and I'll reopen it.

closed 2017-11-07 by flyte