flyte/mqtt-io #13: No such file or directory: 'config.schema.yml'

CWempe, 2017-07-25:
I cannot install and start the program.

During install I get a warning that `MANIFEST.in` has an issue.

```
root@raspi01:~# pip3 install pi-mqtt-gpio
Downloading/unpacking pi-mqtt-gpio
  Downloading pi_mqtt_gpio-0.0.7.tar.gz
  Running setup.py (path:/tmp/pip-build-_j6w1ddf/pi-mqtt-gpio/setup.py) egg_info for package pi-mqtt-gpio
    your setuptools is too old (<12)
    setuptools_scm functionality is degraded
    zip_safe flag not set; analyzing archive contents...

    Installed /tmp/pip-build-_j6w1ddf/pi-mqtt-gpio/pytest_runner-2.11.1-py3.4.egg

    warning: manifest_maker: MANIFEST.in, line 2: 'recursive-include' expects <dir> <pattern1> <pattern2> ...

Downloading/unpacking paho-mqtt (from pi-mqtt-gpio)
  Downloading paho-mqtt-1.3.0.tar.gz (79kB): 79kB downloaded
  Running setup.py (path:/tmp/pip-build-_j6w1ddf/paho-mqtt/setup.py) egg_info for package paho-mqtt
    your setuptools is too old (<12)
    setuptools_scm functionality is degraded
    zip_safe flag not set; analyzing archive contents...

    Installed /tmp/pip-build-_j6w1ddf/paho-mqtt/pytest_runner-2.11.1-py3.4.egg

Downloading/unpacking PyYAML (from pi-mqtt-gpio)
  Downloading PyYAML-3.12.tar.gz (253kB): 253kB downloaded
  Running setup.py (path:/tmp/pip-build-_j6w1ddf/PyYAML/setup.py) egg_info for package PyYAML

Downloading/unpacking enum34 (from pi-mqtt-gpio)
  Downloading enum34-1.1.6-py3-none-any.whl
Downloading/unpacking cerberus (from pi-mqtt-gpio)
  Downloading Cerberus-1.1.tar.gz
  Running setup.py (path:/tmp/pip-build-_j6w1ddf/cerberus/setup.py) egg_info for package cerberus

Installing collected packages: pi-mqtt-gpio, paho-mqtt, PyYAML, enum34, cerberus
  Running setup.py install for pi-mqtt-gpio

    warning: manifest_maker: MANIFEST.in, line 2: 'recursive-include' expects <dir> <pattern1> <pattern2> ...

    Installing pi_mqtt_gpio script to /usr/local/bin
  Running setup.py install for paho-mqtt

  Running setup.py install for PyYAML
    checking if libyaml is compilable
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.4m -c build/temp.linux-armv7l-3.4/check_libyaml.c -o build/temp.linux-armv7l-3.4/check_libyaml.o
    build/temp.linux-armv7l-3.4/check_libyaml.c:2:18: fatal error: yaml.h: Datei oder Verzeichnis nicht gefunden
     #include <yaml.h>
                      ^
    compilation terminated.

    libyaml is not found or a compiler error: forcing --without-libyaml
    (if libyaml is installed correctly, you may need to
     specify the option --include-dirs or uncomment and
     modify the parameter include_dirs in setup.cfg)

  Running setup.py install for cerberus

Successfully installed pi-mqtt-gpio paho-mqtt PyYAML enum34 cerberus
Cleaning up...
```

When I try to start the server I get this:

```
root@raspi01:~# python3 -m pi_mqtt_gpio.server pi_mqtt_gpio.yml
Traceback (most recent call last):
  File "/usr/lib/python3.4/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.4/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.4/dist-packages/pi_mqtt_gpio/server.py", line 32, in <module>
    with open("config.schema.yml") as schema:
FileNotFoundError: [Errno 2] No such file or directory: 'config.schema.yml'
```

I used these commands to install the program:

```
apt-get install python3-pip
pip3 install pi-mqtt-gpio
```

```
Linux raspi01 4.9.0-2-rpi2 #1 SMP Raspbian 4.9.13-1+rpi3 (2017-05-18) armv7l GNU/Linux
```

Any idea what I am doing wrong?

CWempe, 2017-07-26:
I found a workaround to get the program running.

```bash
cd /usr/local/lib/python3.4/dist-packages/pi_mqtt_gpio
wget https://github.com/flyte/pi-mqtt-gpio/raw/develop/config.schema.yml

# now I can run it via:
python3 -m pi_mqtt_gpio.server ~/pi_mqtt_gpio.yml
```

flyte, 2017-07-26:
Ah yes, sorry about that - my fault. Working on a fix now.

flyte, 2017-07-26:
@CWempe Please try with version `0.0.8` and reopen the ticket if the issue persists. Thanks for the bug report.

flyte, 2017-07-26:
Scratch that - still an issue here. I'll fix this in a different way - stand by.

flyte, 2017-07-26:
OK @CWempe please try with version `0.0.9` and reopen if it's still not working for you!

commit 6ae40aa5 (2017-07-26): Include config.schema.yml in package and load it from installed location. Closes #13
closed 2017-07-26 by flyte