flyte/mqtt-io #14: Payload b'ON' does not relate to configured on/off values

CWempe, 2017-07-26:
I cannot send a command via MQTT. :(

It always says `Payload b'*' does not relate to configured on/off values.`

```
root@raspi01:/usr/local/lib/python3.4/dist-packages/pi_mqtt_gpio# python3 -m pi_mqtt_gpio.server ~/pi_mqtt_gpio.example.yml
Module <module 'pi_mqtt_gpio.modules.stdio' from '/usr/local/lib/python3.4/dist-packages/pi_mqtt_gpio/modules/stdio.py'> has no extra requirements to install.
__init__(config={'module': 'stdio', 'name': 'dev'})
setup_pin(pin=21, direction=<PinDirection.OUTPUT: 1>, pullup=None, pin_config={'module': 'dev', 'off_payload': 'OFF', 'pin': 21, 'name': 'test', 'on_payload': 'ON'})
MQTT client: Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
MQTT client: Received CONNACK (0, 0)
Connected to the MQTT broker with protocol v3.1.1.
Subscribed to topic: 'pimqttgpio/mydevice/output/test/set'
Subscribed to topic: 'pimqttgpio/mydevice/output/test/set_on_ms'
Subscribed to topic: 'pimqttgpio/mydevice/output/test/set_off_ms'
MQTT client: Received SUBACK
MQTT client: Received SUBACK
MQTT client: Received SUBACK
Input 'button' state changed to False
MQTT client: Sending PUBLISH (d0, q0, r0, m4), 'b'pimqttgpio/mydevice/input2/button'', ... (3 bytes)
MQTT client: Sending PINGREQ
MQTT client: Received PINGRESP
MQTT client: Sending PINGREQ
MQTT client: Received PINGRESP
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (2 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'ON'
Payload b'ON' does not relate to configured on/off values.
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (2 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'on'
Payload b'on' does not relate to configured on/off values.
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (3 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'off'
Payload b'off' does not relate to configured on/off values.
MQTT client: Sending PINGREQ
MQTT client: Received PINGRESP
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (2 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'ON'
Payload b'ON' does not relate to configured on/off values.
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (2 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'ON'
Payload b'ON' does not relate to configured on/off values.
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (2 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'ON'
Payload b'ON' does not relate to configured on/off values.
MQTT client: Sending PINGREQ
MQTT client: Received PINGRESP
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (2 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'ON'
Payload b'ON' does not relate to configured on/off values.
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (2 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'ON'
Payload b'ON' does not relate to configured on/off values.
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (1 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'1'
Payload b'1' does not relate to configured on/off values.
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (4 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'true'
Payload b'true' does not relate to configured on/off values.
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (4 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'True'
Payload b'True' does not relate to configured on/off values.
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (4 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'TRUE'
Payload b'TRUE' does not relate to configured on/off values.
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'pimqttgpio/mydevice/output/test/set', ...  (1 bytes)
Received message on topic 'pimqttgpio/mydevice/output/test/set': b'1'
Payload b'1' does not relate to configured on/off values.
```

The MQTT messages seem to be fine.

```
pimqttgpio/mydevice/output/test/set ON
pimqttgpio/mydevice/output/test/set on
pimqttgpio/mydevice/output/test/set off
pimqttgpio/mydevice/output/test/set ON
pimqttgpio/mydevice/output/test/set ON
pimqttgpio/mydevice/output/test/set ON
pimqttgpio/mydevice/output/test/set ON
pimqttgpio/mydevice/output/test/set ON
pimqttgpio/mydevice/output/test/set 1
pimqttgpio/mydevice/output/test/set true
pimqttgpio/mydevice/output/test/set True
pimqttgpio/mydevice/output/test/set TRUE
pimqttgpio/mydevice/output/test/set 1
```

Config

```
mqtt:
  host: localhost
  port: 1883
  user: "***"
  password: "****"
  topic_prefix: pimqttgpio/mydevice

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

  - name: dev
    module: stdio

digital_inputs:
  - name: button
    module: raspberrypi
    pin: 22
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes

digital_outputs:
#  - name: bell
#    module: pcf8574
#    pin: 20
#    on_payload: "ON"
#    off_payload: "OFF"

  - name: test
    module: dev
    pin: 21
    on_payload: "ON"
    off_payload: "OFF"
```

Am I doing something wrong?
Why is there a "b" at the beginning of the value?

CWempe, 2017-07-26:
Never mind.
I used the wrong module ``stdio` instead of `raspberrypi`.

CWempe, 2017-07-26:
After #13 and #15 were fixed, I retried this.

The issue occures with `raspberrypi`, too.

**config**

```
mqtt:
  host: localhost
  port: 1883
  user: "***"
  password: "***"
  topic_prefix: test

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: lights
    module: raspberrypi
    pin: 21
    on_payload: "ON"
    off_payload: "OFF"
```

**output**
```
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'test/output/lights/set', ...  (2 bytes)
Received message on topic 'test/output/lights/set': b'UP'
Payload b'UP' does not relate to configured on/off values.
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'test/output/lights/set', ...  (4 bytes)
Received message on topic 'test/output/lights/set': b'down'
Payload b'down' does not relate to configured on/off values.
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'test/output/lights/set', ...  (4 bytes)
Received message on topic 'test/output/lights/set': b'DOWN'
Payload b'DOWN' does not relate to configured on/off values.
```

flyte, 2017-07-26:
Is this using Python 2 or Python 3?

flyte, 2017-07-26:
I believe I know why this happens - the old Python 3 string/bytes mismatch. I'll issue a fix shortly. Thanks for another bug report!

flyte, 2017-07-26:
@CWempe Please could you try with version `0.0.11` and reopen if it's not fixed. Cheers!

flyte, 2017-07-26:
If nothing else, the new log message should provide better information now, if it's not fixed.

closed 2017-07-26 by flyte