flyte/mqtt-io #15: Cannot define digital_outputs with raspberrypi module

CWempe, 2017-07-26:
I cannot get the program started wehen I define `digital_outputs` with `raspberrypi` module.

my config:
```yaml
mqtt:
  host: localhost
  port: 1883
  user: "***"
  password: "*****"
  topic_prefix: test

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: lights
    module: raspberrypi
    pin: 21
    on_payload: "ON"
    off_payload: "OFF"
```

Error:

```
root@raspi01:/usr/local/lib/python3.4/dist-packages/pi_mqtt_gpio# python3 -m pi_mqtt_gpio.server ~/pi_mqtt_gpio_2.yml
Traceback (most recent call last):
  File "/usr/lib/python3.4/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.4/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.4/dist-packages/pi_mqtt_gpio/server.py", line 419, in <module>
    initialise_digital_output(out_conf, GPIO_MODULES[out_conf["module"]])
  File "/usr/local/lib/python3.4/dist-packages/pi_mqtt_gpio/server.py", line 378, in initialise_digital_output
    gpio.setup_pin(out_conf["pin"], PinDirection.OUTPUT, None, out_conf)
  File "/usr/local/lib/python3.4/dist-packages/pi_mqtt_gpio/modules/raspberrypi.py", line 35, in setup_pin
    pullup = PULLUPS[0]
KeyError: 0
```

flyte, 2017-07-26:
Thanks @CWempe - please try version `0.0.10` and reopen the ticket if it's not fixed.

closed 2017-07-26 by flyte

CWempe, 2017-07-26:
It works!

The `b`is still confusing though. 😉 

```
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'test/output/lights/set', ...  (2 bytes)
Received message on topic 'test/output/lights/set': b'ON'
Set 'raspberrypi' output 'lights' to True
MQTT client: Sending PUBLISH (d0, q0, r0, m9), 'b'test/output/lights'', ... (2 bytes)
MQTT client: Received PUBLISH (d0, q0, r0, m0), 'test/output/lights/set', ...  (3 bytes)
Received message on topic 'test/output/lights/set': b'OFF'
Set 'raspberrypi' output 'lights' to False
MQTT client: Sending PUBLISH (d0, q0, r0, m10), 'b'test/output/lights'', ... (3 bytes)
```

flyte, 2017-07-26:
Ah yes, it's the way Python 3 declares that it's a string of bytes with an unknown character encoding. It's useful when debugging, but not necessarily helpful to the end user. I may add some logging configuration which will stop it from being quite so verbose.

Glad the issue's fixed!

commit 370ca003 (2017-07-26): Choose the correct pullup value in raspberrypi module when none is set. Closes #15