flyte/mqtt-io #16: RuntimeWarning: This channel is already in use, continuing anyway

CWempe, 2017-07-26:
When I cancel the program with `Ctrl+C` and restart it, I get the follwoing warning:

```
root@raspi01:~# python3 -m pi_mqtt_gpio.server ~/pi_mqtt_gpio_2.yml
/usr/local/lib/python3.4/dist-packages/pi_mqtt_gpio/modules/raspberrypi.py:39: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self.io.setup(pin, direction, pull_up_down=pullup)
MQTT client: Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
MQTT client: Received CONNACK (0, 0)
Connected to the MQTT broker with protocol v3.1.1.
Subscribed to topic: 'test/output/lights/set'
Subscribed to topic: 'test/output/lights/set_on_ms'
Subscribed to topic: 'test/output/lights/set_off_ms'
```

I do'#t know if this is causing any problem, but maybe it would make sense to include `GPIO.cleanup()`.
see: http://raspi.tv/2013/rpi-gpio-basics-3-how-to-exit-gpio-programs-cleanly-avoid-warnings-and-protect-your-pi

btw: Is there a better way to exit the program?

flyte, 2017-07-26:
You're absolutely right. I've added a cleanup routine to the modules, which is called when the program exits. Please test with version `0.0.12` and reopen if it's not fixed. Thanks!

flyte, 2017-07-26:
ctrl+c is a fine way to exit the program :)

closed 2017-07-26 by flyte

CWempe, 2017-07-26:
It works, too!

Sorry for opening so much issues.
But great work on fixing them so fast. 👍 😎

flyte, 2017-07-26:
> It works, too!

Excellent!

> Sorry for opening so much issues.

Don't be at all. I'm in the process of refactoring and adding more/better tests to the project. I'm also thinking about ways to run the tests on actual Raspberry Pis so I can test the modules properly. Until then, myself and the actual users are the best tests we have! I'd much rather know about the issues so I can make the software better for everybody.

vontrapp, 2018-09-02:
FYI I ran into this error on mine. This is the only gpio program I had run.

I suspect it's when I did a ctrl-c very early in the program, when I started it and immediately remembered something I missed in the config. No big deal, after the next restart the error didn't arise again.

flyte, 2018-09-04:
@vontrapp thanks for reporting it.

If it causes any problems, or you think it shouldn't be this way, then please reopen this issue or create a new one describing how to recreate the problem.