flyte/mqtt-io #17: support LWT (Last Will and Testament)

CWempe, 2017-07-26:
It would be great if the program would support LWT messages.

see: http://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament

and also this comment: http://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/#comment-44936

flyte, 2017-07-26:
Sure, that's a good idea. Does it have to be configurable, or will the following suffice:

- Status topic at `<prefix>/status`
- Publish (retained) on start up: `running`
- Publish (retained) when exiting: `stopped`
- Last will in case of unclean disconnection: `dead`

flyte, 2017-07-26:
I guess it's not a big deal to make it configurable, so I'll do that.

flyte, 2017-07-28:
@CWempe I've added this functionality, so please test with version `0.1.0`. Cheers!

closed 2017-07-28 by flyte

vontrapp, 2018-09-02:
Can LWT be used to set many topics on death? It would be great if I could use that to clear any retained inputs when it is disconnected.

Or maybe there's another mqtt feature that does the same?

flyte, 2018-09-04:
@vontrapp I don't think so. This is a good article on the subject:
https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament

If you want to clear retained messages when it's disconnected, would just setting retain = false work?