flyte/mqtt-io #19: connecting input and output

ssjha, 2017-07-31:
hi,
For my home automation project, i was looking for a way to control a set of relays using mqtt (and Openhab). And as an added feature, i wanted to use a switch array (connected to separate GPIOs on my PI zero)  to control these relays. I tried writing my own python program but failed miserably at it. Recently i came across your implementation and tested it. It is working very well in controlling the relays using mqtt and capturing the button press. However,  i was trying to find a way to control the relay when a button is pressed. 

button 1(GPIO 19) -> "ON"  -> publish mqtt status
|---> pull up (GPIO 4) to control the relay.

One implementation option is subscribe to the mqtt input status message and send a new output message to switch on. This has to be done by Openhab or any other python routine. Is there any other way to achieve this? 

Thanks
Sudhansu

flyte, 2017-07-31:
Hi Sudhansu,

I wrote a whole explanation as to why I don't want to include this in this project, but the more I think about it, the more I'm leaning towards thinking it's a good idea.

**Pros**:
- No reliance on any external service to handle simple local logic
- More responsive

**Cons**:
- Configuration/documentation complication
- Distributed logic - it's much simpler to keep all automation logic to one place (OpenHAB, Home Assistant etc.) so it's easier to debug

**Things to think about**:
- Push to toggle vs direct mapping - do you want to toggle the output on each button press or when the button is off, the output should be off?

Can you think of any other pros/cons/thoughts?

ssjha, 2017-08-01:
Hi Ellis
Thanks for having a detailed thought about the feature. I understand that implementation will have its complication. 

You have listed the pros/cons correctly and I can't think of any additional ones. I believe the primary advantage is with this, the PI module will become a self contained unit and with a small footprint of PI zero/ZeroW, it can be setup as an add-on to the existing controls. 

The button behavior can be configurable as it depends on the type of button used. 
1. Follow the switch (0 for OFF, 1 for ON)
2. Inverse of the switch (1 for OFF, 0 for ON)
3. Toggle (change the state from ON to OFF and vice versa)
This will be inline with the existing functionalities implemented using the inverse and toggle functionality. I am using a push button momentary switch and expect the toggle behavior.

flyte, 2017-08-01:
Something else to think about is the switch debounce. I currently use a
crude method to double check the input but it doesn't scale well if you
have lots of inputs. It would also need to be abstracted and/or have
configurable parameters for each input.

Just dropping this here to remind myself while I'm on my phone.

c3reeman, 2019-02-08:
Was there any change on this ? I too am looking to be able to tie an input to an output or multiple outputs, it would be great for simplicity and avoid home assistant clutter

dandantheflyingman, 2022-01-24:
I was thinking about this too and would a halfway point be the ability to execute a script locally? So say the correct message comes in on a mqtt topic, and it responds by running a script with a set of parameters? 

In a script the user can make it as advanced or as simple as they like.

Does that open any security concerns?

closed 2024-10-17 by BenjiU