flyte/mqtt-io #22: No data Being sent across

BatGirl26, 2017-10-19:
Now, I'm trying to use your Service, but as of yet, I cannot get it to send data across to the Subscribing service.  I have tested the Publish script I have and I am able to Publish and subscribe using the Normal mosquito Client. But your service is not working to send the data across and turn on the GPIO pins. I'm not getting any error messages and it's just sitting there with the message:
MQTT client: Sending PINGREQ
MQTT cleint: Receiving PINGRESP
Now, this is all its doing when I send the message to turn on the GPIO's. I'm new to this, could I please get some tips, please. Is there a step im missing ?

flyte, 2017-10-20:
Hi, could you let me know which topics you're publishing to with what payloads, and paste a copy of your config file please?

flyte, 2017-11-23:
Closing this for now but feel free to open it again with the info requested above.

closed 2017-11-23 by flyte