flyte/mqtt-io #25: Run at boot time + log location?

someinternetguy, 2017-11-22:
Hello, just wanted to say thank you for creating and sharing this program.  Super easy to use and works well.

**_I want to make sure it runs properly at boot time, and I need to be able to access the logs_.**  What is the best way to accomplish this?  Thanks!

flyte, 2017-11-22:
Hey @someinternetguy thanks for saying so! I'm glad it's of use for people.

I tend to use [supervisor](http://supervisord.org/) to run my programs as services. You can tell it to save the output of the program as a log file, and handles the log rotation for you. I should probably add an example deployment really. I'll leave this issue open as a reminder of that.

Hope that helps.

someinternetguy, 2017-11-22:
@flyte, an example would be most helpful... I have used supervisord before but the setup has always come prepackaged.  Anyhow, thanks for the idea... I'll take a look and see if I can figure it out.

someinternetguy, 2017-11-22:
@flyte I was able to get supervisord on raspbian running after monkeying with it for a bit, and I created a .conf file for `pi-mqtt-gpio` under `/etc/supervisor/conf.d/`:
```
[program:mqtt]
command=/usr/bin/python -m pi_mqtt_gpio.server /home/pi/pi-mqtt-gpio-config.yml
autostart=true
autorestart=true
stderr_logfile=/var/log/mqtt-stderr.log
stderr_logfile_maxbytes = 10MB
stderr_logfile_backups = 5
stdout_logfile=/var/log/mqtt-stdout.log
stdout_logfile_maxbytes = 10MB
stdout_logfile_backups = 5
```

However, when I try to run your program using supervisord it fails and I see the following in the stderr log:
```
$ cat /var/log/mqtt-stderr.log
/usr/bin/python: No module named pi_mqtt_gpio
/usr/bin/python: No module named pi_mqtt_gpio
/usr/bin/python: No module named pi_mqtt_gpio
/usr/bin/python: No module named pi_mqtt_gpio
```
I feel like I'm missing the full path for `pi_mqtt_gpio.server` but not sure?

Your program works fine when I run it directly from the command line using `/usr/bin/python -m pi_mqtt_gpio.server /home/pi/pi-mqtt-gpio-config.yml`

Can you post your .conf file for supervisord?

flyte, 2017-11-22:
If you've installed the project into a virtualenv, you need to make sure you're using the right python binary. Change the `command` line to something like:

```
command=/path/to/your/virtualenv/bin/python -m pi_mqtt_gpio.server /home/pi/pi-mqtt-gpio-config.yml
```

I've updated the readme to contain a section about using supervisor. Try setting it up like that if it still doesn't work.

someinternetguy, 2017-11-23:
@flyte I followed your new readme.md supervisor instructions and it worked like a charm.  Many thanks!!!

flyte, 2017-11-23:
Excellent! :)

closed 2017-11-23 by flyte