flyte/mqtt-io #27: feature request

huckpdx, 2017-11-27:
I just discovered your service and am testing it. I see that you recently added the option to configure outputs with retain = true. It would be great to have this option for inputs. 

In my case, I'm using node-red. when I redeploy a flow, the client session is reset, and I don't get input status until it changes state. some of my sensors only change twice a day, so this is a problem for me.

Thanks for considering this request.

flyte, 2017-11-27:
Sounds good to me! I'll add that in this week. Cheers!

flyte, 2017-11-27:
@huckpdx I've implemented this in the `develop` branch. I'm on a train and don't have the ability to test it right now, so I can't release a new version yet. Please test it out from the repository and let me know how it goes.

Cheers!

closed 2017-11-27 by flyte

jfiger, 2019-02-11:
Is the feature "retain inputs" already implemented? Not working for me.

commit 5f3932c7 (2017-11-27): Add option to set retain flag on input value publishes. Closes #27