flyte/mqtt-io #28: Add Banana Pi/Pro Module

iamsven, 2018-01-12:
Hi, this project looks awesome. 
I would love to use it on my Banana Pro board. Unfortunatly I have no Python skills to implement it on my own. Therefore, I would ask you/the community to adapt the raspberry pi module to the wiriningPi library for Banana Pi/Pro 

[Official BPI-WiringPi2-Python library](https://github.com/BPI-SINOVOIP/BPI-WiringPi2-Python)

Thank you very much!

flyte, 2018-01-12:
Hey Sven,

I'd be happy to give it a shot! I don't have the hardware but if you're willing to test it out I can probably knock something together from the docs. 

I'll let you know on here when I've got a bit of time to do it.

Cheers!

flyte, 2018-01-30:
@iamsven Hi!

I've knocked together a module for Banana Pi from the docs. I don't know if it works or not, since I still don't have a board to test with. Would you be able to test it out for me?

You may need to become `root` for this to work, as the Banana Pi docs says it requires `root` privileges.

`sudo su -`

You'll need to clone this repository to your BPi:

`git clone https://github.com/flyte/pi-mqtt-gpio.git`

Check out the feature branch:

```
cd pi-mqtt-gpio
git checkout feature/bananapi
```

Make/adapt your config file, including the new `bananapi` module and pin configuration. Something like this:

```yaml
mqtt:
  host: test.mosquitto.org
  port: 1883
  user: ""
  password: ""
  topic_prefix: pimqttgpio/mydevice

gpio_modules:
  - name: bananapi
    module: bananapi

digital_inputs:
  - name: button
    module: bananapi
    pin: 21
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes

digital_outputs:
  - name: bell
    module: bananapi
    pin: 20
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
```

Create a virtualenv and install requirements:

```
virtualenv ve
. ve/bin/activate
pip install -r requirements.txt
```

Then you should be able to run the server with:

`python -m pi_mqtt_gpio.server`

Let me know how you get on!

iamsven, 2018-01-31:
Hi @flyte ,

this looks good. I'm going to test as soon as possible as possible and post the results below.

iamsven, 2018-02-02:
I installed everything according to your instructions, but when I try to start the server the following error occurs always. I tried it several times and tried to install BPI-WiringPi2-Python manually, but it won't work. 
Did I miss a step?

Error message:
`Last will set on 'test/status' as 'dead'.
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/smartroom/pi_gpio_mqtt/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 439, in <module>
    gpio_config)
  File "/smartroom/pi_gpio_mqtt/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 379, in configure_gpio_module
    install_missing_requirements(gpio_module)
  File "/smartroom/pi_gpio_mqtt/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 211, in install_missing_requirements
    if pkgs_installed.find(pkg_resources.Requirement.parse(req)) is None:
  File "/smartroom/pi_gpio_mqtt/pi-mqtt-gpio/ve/local/lib/python2.7/site-packages/pkg_resources.py", line 2793, in parse
    reqs = list(parse_requirements(s))
  File "/smartroom/pi_gpio_mqtt/pi-mqtt-gpio/ve/local/lib/python2.7/site-packages/pkg_resources.py", line 2721, in parse_requirements
    "version spec")
  File "/smartroom/pi_gpio_mqtt/pi-mqtt-gpio/ve/local/lib/python2.7/site-packages/pkg_resources.py", line 2686, in scan_list
    raise ValueError(msg, line, "at", line[p:])
ValueError: ('Expected version spec in', 'git+https://github.com/BPI-SINOVOIP/BPI-WiringPi2-Python.git', 'at', '+https://github.com/BPI-SINOVOIP/BPI-WiringPi2-Python.git')
`

flyte, 2018-02-04:
Ah, didn't realise it wouldn't cope with installing requirements from git urls. I've fixed that now, so please give it another try when you've got a moment.

Cheers

iamsven, 2018-02-04:
Hi @flyte,

thank you for the quick response and your work. 
This time the installation of [BPI-WiringPi2-Python](https://github.com/BPI-SINOVOIP/BPI-WiringPi2-Python) worked. Unfortunately, there seems to be a problem with BPI-WiringPi2-Python on Banana Pro boards (see error message below). I commented on the [corresponding issue](https://github.com/BPI-SINOVOIP/BPI-WiringPi2-Python/issues/3) on their GitHub, as other people have this problem too. Even though they are stating that the Banana Pro (like Banana Pi M1) is compatible, the issue seems still unfixed (since over a year).

BPI-WiringPi2-Python error message:
```
Successfully installed wiringpi2
Cleaning up...
Unable to determine hardware version. I see: Hardware   : sun7i
,
 - expecting BCM2708 or BCM2709. Please report this to projects@drogon.net
```

I'm sorry, I didn't do this research in advance. 
As it seems that the BPI-WiringPi2-Python team isn't going to release a fix in the next period of time, I would recommend using the official version of RPI.Gpio for LeMaker Banana * boards  ([LeMaker/RPi.GPIO_BP](https://github.com/LeMaker/RPi.GPIO_BP)). This should also work with python.

flyte, 2018-02-11:
@iamsven no worries. Only a minor change required on my end, since I've already got a module for the native Raspberry Pi GPIO.

Please do another `git pull origin feature/bananapi` and change the module name from `bananapi` to `bananapro` in your `config.yml`.

Let me know how it goes.

iamsven, 2018-02-12:
@flyte everything seems to work now!
I tested inputs and outputs on the Banana Pro board, and everything worked as it is supposed to.

Thank you very much for creating a module for Banana Pros.

flyte, 2018-02-12:
Excellent! I'm very glad to hear it :)

iamsven, 2018-04-12:
Hey @flyte,
unfortunately I had to migrate my Banana Pro to the Armbian OS. I tried installing pi-mqtt-gpio the same way as last time, but I always receive the following error message. I tried installing LeMaker/RPi.GPIO_BP outside the virtualenv and the installation was successful. What am I doing wrong?

```
(ve) root@banserv:/opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio# python -m pi_mqtt_gpio.server test.conf
Last will set on '/Smarthome/Relais/status' as 'dead'.
Collecting RPi.GPIO from git+https://github.com/LeMaker/RPi.GPIO_BP@bananapro#egg=RPi.GPIO
  Cloning https://github.com/LeMaker/RPi.GPIO_BP (to bananapro) to /tmp/pip-build-bz07dn1_/RPi.GPIO
Installing collected packages: RPi.GPIO
  Running setup.py install for RPi.GPIO ... error
    Complete output from command /opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/ve/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-bz07dn1_/RPi.GPIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-ijoz79_0-record/install-record.txt --single-version-externally-managed --compile --install-headers /opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/ve/include/site/python3.5/RPi.GPIO:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-3.5
    creating build/lib.linux-armv7l-3.5/RPi
    copying RPi/__init__.py -> build/lib.linux-armv7l-3.5/RPi
    running build_ext
    building 'RPi.GPIO' extension
    creating build/temp.linux-armv7l-3.5
    creating build/temp.linux-armv7l-3.5/source
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-haFsHd/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -I/opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/ve/include/python3.5m -c source/py_gpio.c -o build/temp.linux-armv7l-3.5/source/py_gpio.o
    source/py_gpio.c:23:20: fatal error: Python.h: No such file or directory
     #include "Python.h"
                        ^
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/ve/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-bz07dn1_/RPi.GPIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-ijoz79_0-record/install-record.txt --single-version-externally-managed --compile --install-headers /opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/ve/include/site/python3.5/RPi.GPIO" failed with error code 1 in /tmp/pip-build-bz07dn1_/RPi.GPIO/
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 468, in <module>
    gpio_config)
  File "/opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 408, in configure_gpio_module
    install_missing_requirements(gpio_module)
  File "/opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 250, in install_missing_requirements
    module, pkgs_required))
__main__.CannotInstallModuleRequirements: Unable to install packages for module <module 'pi_mqtt_gpio.modules.bananapro' from '/opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/pi_mqtt_gpio/modules/bananapro.py'> (['git+https://github.com/LeMaker/RPi.GPIO_BP@bananapro#egg=RPi.GPIO'])...

```

flyte, 2018-04-12:
Hey @iamsven 

It looks like you're missing the `python-dev` package (`apt-get install python-dev`) so one of the components isn't able to build.

Try installing that package and running again.

iamsven, 2018-04-14:
I installed the `python-dev ` package. Then I removed the virtual environment and created a new one, following your instructions above. When I run into the following error. I retired it multiple times and did a intense online research, but I couldn't figure out a solution to this...

```
(ve) root@banserv:/opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio# python -m pi_mqtt_gpio.server testc.yml
Last will set on '/Smarthome/Svens-Zimmer/Relais/status' as 'dead'.
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 468, in <module>
    gpio_config)
  File "/opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 408, in configure_gpio_module
    install_missing_requirements(gpio_module)
  File "/opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 243, in install_missing_requirements
    from pip.commands.install import InstallCommand
ImportError: No module named 'pip.commands'

```
Oddly enough, `pip -V` works

```
(ve) root@banserv:/opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio# pip -V
pip 10.0.0 from /opt/smarthome/gpio_mqtt_server/pi-mqtt-gpio/ve/lib/python3.5/site-packages/pip (python 3.5)
```

Do you have any idea for a solution?