flyte/mqtt-io #29: GPIO pull down

ebloved, 2018-01-13:
Hello!
I'm a newbie to python and I can't understand how to pull down input pin.
Even I write pulldown: on in config.yml or edit raspberrypi.py with
```
if pullup is None:
            pullup = PULLUPS[PinPullup.DOWN]
            #pullup = PULLUPS[PinPullup.OFF]
        else:
            pullup = PULLUPS[PinPullup.DOWN]
            #pullup = PULLUPS[pullup]
```
(commented existing settings to not miss a thing) I've got nothing.
Can you help me?
Thanks!

flyte, 2018-01-30:
Hi @ebloved 

You should be able to enable the pulldown by setting it in the config (although you said you already tried):
```yaml
digital_inputs:
  - name: doorbell
    module: raspberrypi
    pin: 22
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes
```

Which module/hardware are you using? Are you able to configure the pulldown using the library itself?

flyte, 2018-04-07:
Going to assume this is sorted. Please feel free to reopen this if you're still having problems.

closed 2018-04-07 by flyte