flyte/mqtt-io #31: Any help would be great...

dawsmac, 2018-04-05:
I have followed the Wiki install and get the following issues 

pi@raspberrypi:~ $ python  pi_mqtt_gpio.server config.yaml
python: can't open file 'pi_mqtt_gpio.server': [Errno 2] No such file or directory

Any help would be great...

flyte, 2018-04-05:
Hi @dawsmac you're missing the `-m` here. The command should be:

```bash
python -m pi_mqtt_gpio.server config.yml
```

Hope that helps.

dawsmac, 2018-04-05:
Hey flyte

Thanks for picking up on this 👍 

Dammit I put the wrong error I had  added the -m and get the following 

pi@raspberrypi:~ $ python -m  pi_mqtt_gpio.server config.yaml
/usr/bin/python: No module named pi_mqtt_gpio
Cheer

Rich

flyte, 2018-04-05:
It looks like the package hasn't been installed properly. Were there any errors from the `pip install pi-mqtt-gpio` command?

dawsmac, 2018-04-06:
Hey Flyte.. 

Cheers for your help... I reflashed the pi and all seems to be working now Thank again... I have married this up with my https://www.home-assistant.io/  linking up all my PIR and Door sensors around the house :)!!!

closed 2018-04-06 by dawsmac

flyte, 2018-04-06:
Great! Glad to hear it's working for you :)