flyte/mqtt-io #32: DHT11 or DHT22 sensor

dawsmac, 2018-04-07:
Hi 
If this is a pull request sorry.. 

I have a few DHT11 sensors and was hoping I could add them to this setup, is it possible? 

Cheers
Rich

flyte, 2018-04-07:
Hmm, that would be handy, wouldn't it? I'll have a think about it and see if I can include it next time I work on the project.

dawsmac, 2018-04-08:
Cool sounds good....

flyte, 2018-08-15:
Reopening to keep the feature request around for if I ever get some time off to work on it.

eXCepT74, 2018-10-25:
Is that possible to add support for MCP23017 - it has 16 I/O ports , I2C and each port can be loaded with 20mA , it has as well switchable active pullup resistors. I probably will use ur project to control my smart house based on PCF, but MCP is good idea as well. For now i see that will need at least two expanders on pcf... If i use two expanders with same addresses whether those two will be work same - i mean - if ill connect leds to both - the states of leds will be same or on i2c will be conflict due to same addresses ? i need one pcf to connect to control lamps ( leds )for my house. Anyway great project! Keep going like that!!! regards

flyte, 2018-10-26:
Hey @eXCepT74

The MCP23017 is certainly suitable for use with this project, but could you make a separate Issue please? You can just copy and paste your comment into a new one.

Cheers

eXCepT74, 2018-11-05:
-

patvdleer, 2019-02-18:
Following :)

BenjiU, 2019-02-18:
Currently I'm working on the interrupt issue (#5), but how may a yaml config for dht11/dht22 look? You have two values on one sensor, that should appear in two different topics. My suggestion (I steel the config idea as I've implemented it for lm75 i2c temperatur sensor in pullrequest #52)

```
sensor_modules:             //list for all sensor_modules, like for gpio_modules
  - name: dht11_sensor_temperature
    module: dht11
    gpio: 16                        //pin dht11 sensor is connected to
    value: temperatur          
  - name: dht11_sensor_humidity
    module: dht11
    gpio: 16                        //pin dht11 sensor is connected to
    value: humidity
```
So we config the dht11/22 sensor twice on the same pin. First time with temperature value and second time with humidity value. These values will be passed to module instance and stored intern, to select which value should be returned.

In the sensor_inputs we create two entries, too, first using the dht11_sensor_temperature and second using dht11_sensor_humidity. This gives us the possibility to configure the mqtt topic and for example different update times:
```
sensor_inputs:
  - name: my_temperature              // mqtt topic
    module: dht11_sensor_temperature
    interval: 120                                // 2*60sec, each 2 min
    digits: 2                                       // digits to round the value to
  - name: my_humidity
    module: dht11_sensor_humidity
    interval: 240                               // e.g. every 4 minutes
    digits: 1
```
This should create the following mqtt topics with values:
```
mydevice/sensor/my_temperature 19.50
mydevice/sensor/my_temperature 19.50
mydevice/sensor/my_humidity 41.0
mydevice/sensor/my_temperature 19.50
mydevice/sensor/my_temperature 19.50
mydevice/sensor/my_humidity 41.0
```
Any suggestions or does someone see a problem?

flyte, 2019-02-18:
It would be nice to keep the module configs 1:1 with the physical modules, so in `sensor_modules` it should be listed once. The module name doesn't have a bearing on the topic, so the temperature/humidity configuration could be on the `sensor_inputs` instead. Something like:

```yaml
sensor_modules:
  - name: dht11_sensor
    module: dht11
    i2c_bus_num: 1
    chip_addr: 0x20

sensor_inputs:
  - name: livingroom_temp
    module: dht11_sensor
    interval: 120
    digits: 2
    sensor: temperature
  - name: livingroom_humidity
    module: dht11_sensor
    interval: 240
    digits: 1
    sensor: humidity
```

What do you reckon?

BenjiU, 2019-02-19:
That's of course the better solution. I already implemented this "interface" for LM75 I2C sensor on the other branch: when getting the value from the sensor in the `sensor_timer_thread`, the `sensor` is passed to the read function. We just need to pass the `sens_conf` to the `get_value()` function, instead of the `sensor` value:
```
                sensor = SENSOR_MODULES[sens_conf["module"]]
                try:
                    value = round(sensor.get_value(sensor),
                                  sens_conf["digits"])
```
the `get_value` function can now evaluate the `sensor` attribute and return the value.

I have two other points:
1) Does someone know a good, free, usable DHT11/22 library?
2) I read, that the DHT11/22 sensor shouldn't be read more often then every 1 or 2 seconds. This should be kept in mind for implementation: if we read the temperature with the first `get_value()` call, and directly after this the humidity with the second `get_value()`, the DHT sensor class must recognize this and return the previous read value, and is not allowed to read the DHT11/22 again within this short time.

patvdleer, 2019-02-19:
Would this work for a ds18b20 as well?

BenjiU, 2019-02-20:
> Would this work for a ds18b20 as well?
No, as far as I know, one wire from dht11 and one wire from ds18b20 are incompatible. So you should create a new issue for ds18b20, if you want it to be supported. Maybe you could try a library, that should be supported and mention it there. (e.g. https://github.com/timofurrer/w1thermsensor)

gieljnssns, 2019-04-05:
@BenjiU 
> Does someone know a good, free, usable DHT11/22 library?

Maybe this one?
https://github.com/adafruit/Adafruit_Python_DHT

julioo94, 2019-07-27:
Hello,

I blame you because I try to set up pi-mqtt-gpio with DHT22

I already have gpio switch set up and it works very well

mes as I add the DHT22 and I run the program here is what I put I install the library adafruit

 Config did not validate:
sensor_inputs:
- 0:
  - type:
    - unknown field
  1:
  - type:
    - unknown field

if one of you to an idea to help me on the way is DHT22 it will be great :)

oguven, 2019-09-03:
Hello,

I'm also having the same problem with DHT11.

Installed pi-mqtt-sensor and Adafruit_Python_DHT with pip (sudo privileges)

My config file works without sensor lines. When I add DHT sensor lines;

`Config did not validate:
sensor_inputs:
- 0:
  - type:
    - unknown field`

here my config.yml

`mqtt:
  host: (webaddress)
  port: (port)
  tls:
    enabled: yes
    cert_reqs: CERT_NONE
    insecure: yes
  user: "(user)"
  password: "(pass)"
  topic_prefix: station
  status_topic: status
  status_payload_running: running
  status_payload_stopped: stopped
  status_payload_dead: dead

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

sensor_modules:
  - name: dht22
    module: dht22
    type: dht11
    pin: 5

sensor_inputs:
  - name: dht22_temperature
    module: dht22
    interval: 10
    digits: 4
    type: temperature`

flyte, 2021-01-24:
I believe this is fixed now. Please feel free to comment if it's not.

closed 2021-01-24 by flyte