flyte/mqtt-io #33: Install trouble

ronlandrus, 2018-04-27:
pi-mqtt-gpio is rolling reboot. see log snapshot below. any suggestions? thanks.

Last will set on 'home/garage/status' as 'dead'.
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 449, in <module>
    gpio_config)
  File "/home/pi/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 389, in configure_gpio_module
    install_missing_requirements(gpio_module)
  File "/home/pi/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 215, in install_missing_requirements
    from pip.commands.install import InstallCommand
ImportError: No module named commands.install

flyte, 2018-04-30:
Hi @ronlandrus.

It looks like maybe you've got an old version of virtualenv and pip. Could you try upgrading pip within your virtualenv and trying again?

```bash
pip install --upgrade pip
```

darkpainy, 2018-04-30:
Hi @flyte  

I have the same issue, running pip install --upgrade pip gives me: Requirement already up-to-date: pip in ./ve/lib/python2.7/site-packages (10.0.1)

Thanks in advance for any help you can offer :)

darkpainy, 2018-04-30:
hi @ronlandrus , @flyte 

I solved my problem by running as python3.5

changing this: 
[program:pi_mqtt_gpio]
command = /home/pi/ve/bin/python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
directory = /home/pi
redirect_stderr = true
stdout_logfile = /var/log/pi-mqtt-gpio.log

to this:
[program:pi_mqtt_gpio]
command = python3.5 -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
directory = /home/pi
redirect_stderr = true
stdout_logfile = /var/log/pi-mqtt-gpio.log

Maybe this is not a good solution, but it works for me, nothing else did.

flyte, 2018-05-01:
Interesting. Running in Python 3 should be fine. I'll make a note to double check that everything still works how it should in Python 2.

@ronlandrus does that work for you?

ronlandrus, 2018-05-05:
Not better just different halt with fatal log below
/usr/bin/python3.5: Error while finding module specification for 'pi_mqtt_gpio.server' (ImportError: No module named 'pi_mqtt_gpio')
also pi@raspZero1:~/ve/lib/python2.7/site-packages
Maybe by running with python3.5 it has issue finding site-packages.
Is it a path issue?
Maybe I should do fresh install and use python3 to install pip etc.

flyte, 2018-05-05:
Yeah, packages are installed separately between the two python versions.
You'll have to do the process again for python 3.

ronlandrus, 2018-05-06:
tried installing all to use python3.5 and added some related PATH's to ./profile but am still getting rolling reboot of pi-mqtt_gpio. Log below

pi@rasZero1:~ $ sudo supervisorctl status
pi_mqtt_gpio                     FATAL     Exited too quickly (process log may have details)

pi@rasZero1:~ $ cat /var/log/pi-mqtt-gpio.log
/usr/bin/python3.5: Error while finding module specification for 'pi_mqtt_gpio.server' (ImportError: No module named 'pi_mqtt_gpio')

.profile
PATH=$PATH:/home/pi/.local/lib/python3.5/site-packages
PATH=$PATH:/home/pi/.local/lib/python3.5
PATH=$PATH:/home/pi/.local/lib/python2.7/site-packages
ATH=$PATH:/home/pi/.local/lib/python2.7
PATH=$PATH:/home/pi/ve/bin
PATH=$PATH:/usr/bin/pip3
PATH=$PATH:/home/pi/ve/bin/python3
PATH=$PATH:/home/pi/ve/bin/python

darkpainy, 2018-05-06:
@ronlandrus try using pip3 to install the pi_mqtt_gpio package. Otherwise python 3+ won't find your install

ronlandrus, 2018-05-06:
Unfortunately I tried that.
(ve) pi@rasZero1:~ $ pip3 uninstall pi-mqtt-gpio

ronlandrus, 2018-05-12:
Had to start fresh with older Raspbian. But appears to function, have not fully tested.
This is going on my garage door hopefully.

MQTT client: Received SUBACK
MQTT client: Received PUBACK (Mid: 9)
MQTT client: Sending PINGREQ
MQTT client: Received PINGRESP

pi@rasZero1:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"

pi@rasZero1:~ $ python3 --version
Python 3.4.2

ronlandrus, 2018-05-18:
Oh yeah.
pi@rasZero1:~ $ pip --version
pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)
pi@rasZero1:~ $ pip3 --version
pip 1.5.6 from /usr/lib/python3/dist-packages (python 3.4)
pi@rasZero1:~ $

CWempe, 2018-07-04:
Any update on this?

I have the same ussue.

I did a fresh install on a raspi.

```
(ve) cwempe@raspi2a:~$ python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
Last will set on 'pimqttgpio/mydevice/status' as 'dead'.
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/cwempe/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 449, in <module>
    gpio_config)
  File "/home/cwempe/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 389, in configure_gpio_module
    install_missing_requirements(gpio_module)
  File "/home/cwempe/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 215, in install_missing_requirements
    from pip.commands.install import InstallCommand
ImportError: No module named commands.install
```

I tried using `pip<10.0.0`.

```
(ve) cwempe@raspi2a:~$ pip --version
pip 9.0.3 from /home/cwempe/ve/local/lib/python2.7/site-packages (python 2.7)

(ve) cwempe@raspi2a:~$ python --version
Python 2.7.9
```

But I still cannot start the tool successfully. 😞 

```
(ve) cwempe@raspi2a:~$ python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
Last will set on 'pimqttgpio/mydevice/status' as 'dead'.
Collecting pcf8574
  Downloading https://files.pythonhosted.org/packages/5f/ab/c8955e73d866977c221fafc167a2fa6bdab5f07b3569fb2dec1d8a7a8836/pcf8574-0.0.5.tar.gz
Collecting smbus-cffi (from pcf8574)
  Downloading https://files.pythonhosted.org/packages/29/3d/a50bd997979c7554c9c571753d34e976eb88ebf41d3a66accb2468bd3c69/smbus-cffi-0.5.1.tar.gz
    Complete output from command python setup.py egg_info:
    c/_cffi_backend.c:2:20: fatal error: Python.h: Datei oder Verzeichnis nicht gefunden
     #include <Python.h>
                        ^
    compilation terminated.
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-p1En51/smbus-cffi/setup.py", line 48, in <module>
        'Topic :: System :: Hardware',
      File "/home/cwempe/ve/local/lib/python2.7/site-packages/setuptools/__init__.py", line 128, in setup
        _install_setup_requires(attrs)
      File "/home/cwempe/ve/local/lib/python2.7/site-packages/setuptools/__init__.py", line 123, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/home/cwempe/ve/local/lib/python2.7/site-packages/setuptools/dist.py", line 514, in fetch_build_eggs
        replace_conflicting=True,
      File "/home/cwempe/ve/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 770, in resolve
        replace_conflicting=replace_conflicting
      File "/home/cwempe/ve/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1053, in best_match
        return self.obtain(req, installer)
      File "/home/cwempe/ve/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1065, in obtain
        return installer(requirement)
      File "/home/cwempe/ve/local/lib/python2.7/site-packages/setuptools/dist.py", line 581, in fetch_build_egg
        return cmd.easy_install(req)
      File "/home/cwempe/ve/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 673, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/home/cwempe/ve/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 699, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/home/cwempe/ve/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 884, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/home/cwempe/ve/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1152, in build_and_install
        self.run_setup(setup_script, setup_base, args)
      File "/home/cwempe/ve/local/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1140, in run_setup
        raise DistutilsError("Setup script exited with %s" % (v.args[0],))
    distutils.errors.DistutilsError: Setup script exited with error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-p1En51/smbus-cffi/
You are using pip version 9.0.3, however version 10.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/cwempe/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 449, in <module>
    gpio_config)
  File "/home/cwempe/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 389, in configure_gpio_module
    install_missing_requirements(gpio_module)
  File "/home/cwempe/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 222, in install_missing_requirements
    module, pkgs_required))
__main__.CannotInstallModuleRequirements: Unable to install packages for module <module 'pi_mqtt_gpio.modules.pcf8574' from '/home/cwempe/ve/local/lib/python2.7/site-packages/pi_mqtt_gpio/modules/pcf8574.pyc'> (['pcf8574'])...
```

Other sites say installing `python-dev` would fix this.
But I am not sure how this works inside `virtualenv`. 🤔 

` sudo apt-get install python-dev` did not help.

flyte, 2018-07-05:
Hey @CWempe sorry, I haven't got around to it yet.

Installing `python-dev` won't fix it, you'll have to use the workaround I posted before, which is to downgrade pip to < 10.0.0:
```
pip install --upgrade "pip<10.0.0"
```

This should also work inside a virtualenv.

flyte, 2018-07-05:
Hmm, that's odd. Installing `python-dev` should have fixed that. Maybe try recreating the virtualenv now that `python-dev` is installed?

Sorry, I haven't got around to fixing this yet, but I'll make it a priority.

flyte, 2018-07-05:
Ah, so now you're getting a different error. It can't find `ffi.h` this time.

Try `sudo apt-get install libffi-dev` and running it again.

CWempe, 2018-07-05:
`pip install pcf8574` worked this time.

Now I get this error.
```
(ve) cwempe@raspi2a:~$ python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
Last will set on 'pimqttgpio/mydevice/status' as 'dead'.
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/cwempe/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 449, in <module>
    gpio_config)
  File "/home/cwempe/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 390, in configure_gpio_module
    return gpio_module.GPIO(gpio_config)
  File "/home/cwempe/ve/local/lib/python2.7/site-packages/pi_mqtt_gpio/modules/pcf8574.py", line 34, in __init__
    self.io = PCF8574(config["i2c_bus_num"], config["chip_addr"])
  File "/home/cwempe/ve/local/lib/python2.7/site-packages/pcf8574/__init__.py", line 52, in __init__
    self.bus = smbus.SMBus(i2c_bus_no)
  File "/home/cwempe/ve/local/lib/python2.7/site-packages/smbus/smbus.py", line 56, in __init__
    self.open(bus)
  File "/home/cwempe/ve/local/lib/python2.7/site-packages/smbus/smbus.py", line 83, in open
    raise IOError(e.errno)
IOError: 2
```

In case my `pi-mqtt-gpio.yml` is the problem now:

```yaml
mqtt:
  host: mosquitto
  port: 1883
  user: openhab
  password: **************
  topic_prefix: pimqttgpio/mydevice

gpio_modules:
  - name: pcf8574i
    module: pcf8574
    i2c_bus_num: 1
    chip_addr: 0x20

  - name: pcf8574o
    module: pcf8574
    i2c_bus_num: 1
    chip_addr: 0x38

digital_inputs:
  - name: button
    module: pcf8574i
    pin: 2
    on_payload: "ON"
    off_payload: "OFF"

digital_outputs:
  - name: bell
    module: pcf8574o
    pin: 2
    on_payload: "ON"
    off_payload: "OFF"
```

flyte, 2018-07-06:
Now it looks like it's happily installed all of the requirements, but it's not able to open the I2C device to communicate with the pcf8574. I believe you'll need to add your user to the `i2c` group in order to do this:

```
sudo adduser -aG i2c cwempe
```

Log out and log back in again, then it'll hopefully work (or provide the next error message!). Alternatively, run as root, but adding your user to the right groups is the preferable solution.

CWempe, 2018-07-06:
It is working now.

I will post a summary soon.

Thank you very much! 👍

CWempe, 2018-07-06:
Because I had installed my raspi with [raspbian-ua-netinst](https://github.com/debian-pi/raspbian-ua-netinst) and some customization, I had to install and configure some things extra.

# Enable I2C

```
sudo apt-get update
sudo apt-get install raspi-config i2c-tools

# enable i2c interface
sudo raspi-config

sudo reboot

# add my user to `i2c` group
sudo adduser cwempe i2c

# detect connected i2c devices
sudo i2cdetect -y 1
```

# Install missing packages.

```
sudo apt-get install python-dev libffi-dev python-pip
```

# Create and enter virtual enviroment

```
sudo pip install --upgrade virtualenv
cd ~
virtualenv ve
. ve/bin/activate
```
# Downgrade pip and install pi-mqtt-gpio

```
pip install --upgrade "pip<10.0.0"
pip install pi-mqtt-gpio
```

# Create configuration files

See `README`

# Test

```
/home/cwempe/ve/bin/python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
```

flyte, 2018-07-09:
@CWempe great! Thank you very much for the summary. I'll try to incorporate some of it into the readme when I get some time.

flyte, 2018-08-16:
Finally fixed in v0.2.5! Thanks for the report @ronlandrus. Sorry it took so long. This should now work with all of the latest versions of pip.

closed 2018-08-16 by flyte