flyte/mqtt-io #36: Getting FATAL ERROR Message

FreeDal, 2018-08-20:
Hi Flyte,

Thanks for the hard work on this. I installed the script last week and had no issues at all. Today I'm getting the following error. 

pi@pi:~ $ pip install pi-mqtt-gpio
...
Successfully installed PyYAML-3.13 cerberus-1.2 enum34-1.1.6 paho-mqtt-1.3.1 pi-mqtt-gpio-0.2.5

pi@pi:~ $ python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml 
FATAL ERROR: The file at pi_mqtt_gpio/__init__.py should be replaced using 'make schema' before packaging.

I'm using the same Rasberry Pi 2B v1.1 as I used last week. Please let me know if you need any other information.

I'm grateful for any advice.

Thanks,
FreeDal.

flyte, 2018-08-20:
Argh, sorry, my bad.

Thank you for reporting this!

I'm just building a new version now (0.2.6) which will fix that. I'll reply again shortly once it's done.

flyte, 2018-08-20:
@FreeDal please try installing again now.

flyte, 2018-08-20:
Excellent! Glad to hear it.

closed 2018-08-20 by flyte