flyte/mqtt-io #38: Momentary output

vontrapp, 2018-09-02:
There are cases where you absolutely want an output to be on (off) for a short period. It would be nice to have this contained in the raspi config and not count on a remote client sending the correct temporary command.

flyte, 2018-09-04:
Hi @vontrapp 

I think this is a duplicate of #10 which I've already implemented.

You can use the `set_on_ms` and `set_off_ms` topic suffixes to do this.

Please see this readme section for more info:
https://github.com/flyte/pi-mqtt-gpio#temporary-set

flyte, 2018-09-04:
Oh, hang on - now I see what you mean. You'd like to use the `/set` suffix but have the momentary part configured in the config file. Hmm. I'll have a think about this!

vontrapp, 2018-09-05:
And I don't think this would be a replacement for the "set_on_ms" either, as sometimes you do want that flexibility to do different durations or entirely switch on or off etc.

flyte, 2018-09-05:
Yes, I absolutely agree with you. OK, I'll have a look at how this can be
done.

reestr, 2019-08-06:
I have the same use case as @vontrapp has outlined.  I would like this for safe opening of garage door and to switch my outside lights which are on a relay circuit on/off.

iondarie, 2020-01-01:
any news on this?

flyte, 2020-01-02:
@vontrapp @reestr @iondarie I've implemented this and updated the readme on the `develop` branch. Can you guys test it out before I release it? It's the new `timed_set_ms` digital output config option. For example, if you want to set the output for one second, set `timed_set_ms: 1000`.

iondarie, 2020-01-02:
Sure i will try in a few hours when i get back home.

Thank you

iondarie, 2020-01-02:
TESTED and it works perfect! thank you

flyte, 2020-01-03:
OK great. Thanks for testing it. I'll release a new version in a little while.

closed 2021-01-24 by flyte