flyte/mqtt-io #45: PCF8574 & server crash

eXCepT74, 2018-11-12:
Hello . 

I just got some pcf expanders and if setup config for pcf8574 as output - then is all right , but same script with setup as input just one single pin crashing that app. And by the way - can i use in one pcf some pins as input and some as output ?

There is my test config:

mqtt:
  host: 192.168.0.251
  port: 1883
  user: ""
  password: ""
  topic_prefix: pimqttgpio/mydevice

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

  - name: pcf8574
    module: pcf8574
    i2c_bus_num: 1
    chip_addr: 0x38

digital_inputs:
  - name: button
    module: raspberrypi
    pin: 21
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes

  - name: testPCF8574
    module: pcf8574
    pin: 1  # This device is connected to pin 21 of the Raspberry PI GPIO
    on_payload: "ON"
    off_payload: "OFF"

And here is error log:

pi@HomeBridge:~ $ python -m pi_mqtt_gpio.server org_config.yml 
Last will set on 'pimqttgpio/mydevice/status' as 'dead'.
MQTT client: Sending CONNECT (u0, p0, wr1, wq1, wf1, c0, k60) client_id=pi-mqtt-gpio-429373a40b854779a60eae64cd1cef6dbaf7456f
Input 'button' state changed to False
MQTT client: Sending PUBLISH (d0, q0, r0, m1), 'pimqttgpio/mydevice/input/button', ... (3 bytes)
MQTT client: Sending PUBLISH (d0, q1, r1, m2), 'pimqttgpio/mydevice/status', ... (7 bytes)
MQTT client: Sending DISCONNECT
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 481, in <module>
    state = bool(gpio.get_pin(in_conf["pin"]))
  File "/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/modules/pcf8574.py", line 50, in get_pin
    return self.io.port[pin]
  File "/home/pi/.local/lib/python2.7/site-packages/pcf8574/__init__.py", line 22, in __getitem__
    return self.pcf8574.get_pin_state(key)
  File "/home/pi/.local/lib/python2.7/site-packages/pcf8574/__init__.py", line 93, in get_pin_state
    state = self.bus.read_byte(self.address)
  File "/home/pi/.local/lib/python2.7/site-packages/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
  File "/home/pi/.local/lib/python2.7/site-packages/smbus/smbus.py", line 110, in read_byte
    raise IOError(ffi.errno)
IOError: 5

-----------------------------------------------------------------------------------------------------

There is enough to set that pcf's pin as output and error is gone. 
But i need 3x pcf extenders set as input...  My config is simple so far cause i wanna to check simple config at first with my homekit2mqtt homekit bridge.

Any idea whats wrong ?

flyte, 2018-11-13:
That was easy :)

flyte, 2018-11-13:
Out of interest though, what was the solution?

closed 2018-11-13 by flyte

eXCepT74, 2018-11-14:
The problem was server crashing when was used PCF8574 extender. I wrote config with multiple inputs. And after start server crashed. Then for find the problem - i removed all inputs except one - restart and server crashed :( .
So i changed inputs to outputs - restart and all was ok. In one word  - if server worked as inputs never started just crashed with many lines - here on this page was full log for 15 minutes ;).
Finally i solved problem - some i2c component from rpi3 was hang i think. Restart solved problem.
I dont wanted to bother you. I doing full smarthome project on your code. I use as well Tasmota firmware and homekit2MQTT. I will use manu inputs and outputs by GPIOs and PCF expanders that i think we will speak many times ;) .  I see that you already like it ;).
Ok no more jokes. Good job mate !