flyte/mqtt-io #47: Is ist possible to change gpio.BCM to gpi.BOARD for Raspberry module

Rohrleitner, 2018-12-11:
Hi! Your script seems nearly perfect! Thank you.
But is ist possible to change gpio.BCM to gpi.BOARD for Raspberry module.
I found it hardcoded in line 54 from "raspberrypi.py" "        gpio.setmode(gpio.BCM)"

Greetings from austria
Charly

Rohrleitner, 2018-12-12:
> Schön abend, Charly!
>
> I believe its possible, yes. I'll take a look at the code later, but from
> memory I expect it to be a simple change.
>
> Ich werde versuchen es machen bald.

Perfect!
Is there an information about, how to use the "timeoput" output with the
"ms" value

BenjiU, 2019-02-19:
What would be the default mode? GPIO.BCM?
Should we check for errors, if another python module changed the numbering and we could get a inconsistancy?

To detect which pin numbering system has been set (for example, by another Python module):
mode = GPIO.getmode()
The mode will be GPIO.BOARD, GPIO.BCM or None

flyte, 2019-05-26:
Good questions, @BenjiU 

I'd say use what we're using now for numbering as default, if only to be more backwards compatible. Is one more of a defacto standard than the other?

When would you suggest checking the mode? If we're setting the mode explicitly, then that should suffice during the setup phase, but do you suggest checking the mode before each read/write operation too? Perhaps this should be configurable, since it may add a performance penalty.

BenjiU, 2019-05-27:
Hi,
I'm always using gpio.BCM, keep it for backward compatibility. Good idea.

I just googled a little bit and I'm not sure, if we need a check. Does someone know, how it works, when linux program uses a library, is it copied and exclusive or shared with different executed programs? Maybe we don't need a check. I'll try next week on a raspberry with different programs.

yozik04, 2020-01-23:
@BenjiU Did you had a chance to check? I have implemented that for Orange Pi in Orange Pi pull request.