flyte/mqtt-io #48: Add bouncetime? Problems with Bouncing on digital input!

Rohrleitner, 2018-12-13:
Hello. 
My first garagdoor automation setup works really fine. 
The only problem is, that i have a "bouncing", everytime from "CLOSED-OPEN-CLOSED" on my reed switchs and i can see it switching in your log.

There should be a function for setting the bouncetime:
[https://sourceforge.net/p/raspberry-gpio-python/wiki/Inputs/](url)

# add rising edge detection on a channel, ignoring further edges for 200ms for switch bounce handling
GPIO.add_event_detect(channel, GPIO.RISING, callback=my_callback, bouncetime=200)

Ist it possible to set the bouncetime for each digital input in the yml?
e.g. bouncetime: 200 in the yml?
and if bouncetime >0 then run GPIO.add_event_detect?

Would be a great help if you could bring this in!

Greetings charly

flyte, 2019-01-29:
Hi @Rohrleitner 

Yes, this looks like a good idea. Thanks. I'll add this to the list and try to get it implemented soon.

Cheers,
Ellis

bigbigblue, 2019-06-13:
I would very much like to see this added too. At the moment I am looking at using hardware debouncing of PIR contacts attached to my Pi.

pixiandreas, 2019-06-30:
Hi there, lovely software, any news about this issue ??

Regards

Andreas

BenjiU, 2019-07-01:
Hi,

I'm working on that, and have already a working version, but I have to clean up a lot of code :-( before I can create a pull request.
Currently I have extended the digital_inputs config (https://github.com/BenjiU/pi-mqtt-gpio/commit/96bf4e11cd25b2d8e9d90113d8ad05959de1b846#diff-0ab99ddee45bf345e48d79ae48ea7ecd)
with a interrupt-type (if none, its polling)
      interrupt:
        type: string
        required: no
        default: none
        allowed:
          - rising
          - falling
          - both
          - none
      bouncetime:
        type: integer
        required: no
        default: 100
        min: 1
and a bouncetime, that may be configured (default is 100ms).

Would that fit your needs?
There are only interrupts for RaspberryPi, not for attachable modules yet.

pixiandreas, 2019-08-29:
Hi

Sorry missed your answer.

I'm interested  to test tis for you.

Andreas

iondarie, 2020-01-02:
I have the same problem how can i test?

digital_inputs:
  - name: mare_closed
    module: raspberrypi
    pin: 25
    on_payload: "TRUE"
    off_payload: "FALSE"
    pullup: no
    pulldown: no
    retain: yes