flyte/mqtt-io #49: Two piface2 modules on same board

XolphinMaarten, 2019-01-21:
Would it be possible to add support for two piface2 modules? At the moment only the first piface2 seems to be supported, and there does not seem to be an option to select the second one on a different address. With the address jumpers 4 modules in total are allowed on the same Pi.

I would like something like this:

```
gpio_modules:
  - name: piface2-1
    module: piface2
    chip_select: 0

  - name: piface2-2
    module: piface2
    chip_select: 1
```

flyte, 2019-01-29:
Hi @XolphinMaarten 

Ah yes, I didn't consider this before. This is a good idea, and I'll add it to the list to implement once I get around to spending some time on this project again.

Cheers,
Ellis

szimszon, 2022-02-03:
Hi,

Any news on this?