flyte/mqtt-io #53: Add support for MCP23017 16 bit I/O expander

flyte, 2019-02-18:
Originally posted as a pull request by @eXCepT74 so I've moved it to an issue:

> Is that possible to add support for MCP23017 - it has 16 I/O ports , I2C and each port can be loaded with 20mA , it has as well switchable active pullup resistors. I probably will use ur project to control my smart house based on PCF, but MCP is good idea as well. Anyway great project! Keep going like that!!! regards

rbswift, 2019-09-08:
Yes please!

This would be great because then the AB Electronics IO Pi Plus 32 channel IO expansion board would be supported. (https://www.abelectronics.co.uk/p/54/io-pi-plus).

I see the MCP23017 is the I2C version of the MCP23S17 which uses SPI and is used on the piface2 expansion board which is already supported.

I'm currently using piface2 successfully but need more IOs

rbswift, 2019-09-08:
there is actually some sample code here that might be useful...

https://github.com/abelectronicsuk/ABElectronics_Python_Libraries

genex89, 2019-09-11:
I Need support for MCP23017 - it has 16 I/O ports. News about this?

Misiu, 2019-10-15:
Home Assistant recently added MCP23017 support - https://github.com/home-assistant/home-assistant/pull/23127 maybe this code will be useful

rbswift, 2020-04-19:
Is there any update on this? It seems like quite a popular chip. I'd really like to have 16 I/Os

aerodolphin, 2020-10-13:
Any update on this? I came from openhab, and I would like to use my expander boards That I develop with MCP23017. 

I have some boards with 48 I/O and another boards with 32 I/O. It's very useful for everyone like me that want to have a lot of wired sensors and switches.

aerodolphin, 2020-10-16:
Can someone give some guidance to develop this new module or make it? I can pay for the job.

flyte, 2020-10-16:
Hey, sorry for the delay. Is there an existing Python MCP23016 library that we could use for integration? If so, it's a relatively simple job of copying one of the other modules for reference and modifying it to work with the different library.

flyte, 2020-10-16:
I've just read the rest of the comments.. Ideally we'd use some kind of pip-installable package that enables control of the MCP23017. I suspect there is something usable, and I'll have a look this weekend.

aerodolphin, 2020-10-16:
The is some Python libraries on the internet. Here you have two examples:

[https://pypi.org/project/RPi-MCP23017/](https://pypi.org/project/RPi-MCP23017/)

[https://github.com/owainm713/MCP230XX-Python-Module](https://github.com/owainm713/MCP230XX-Python-Module)

Maybe can be useful. Thanks for the help.

I will share after the hardware info that can be used by anyone.

aerodolphin, 2020-10-19:
Here is a photo of the 48Ch expander that I develop to expand the Raspberry Pi GPIO IOs by I2C with MCP23017 multiplexers.

[image]

Misiu, 2020-10-19:
@aerodolphin this looks awesome 👍🎉🚀
Are you planning to release the schematics? I'd like to make one to use this with Home Assistant (it supports MCP23017)
I don't need 48 I/O, 30 will be enough, but maybe in the future, they will be used.

aerodolphin, 2020-10-19:
@Misiu Yes, I will release the schematics and also can sell the PCB board and components assembled or in kit. As soon as finish all the tests.

Misiu, 2020-10-19:
@aerodolphin I need this for a school project, so I'll try to build one on my own or buy it if the price will be good (sadly a low budget), but please let me know when tests are over. Really looking forward to this!

flyte, 2020-10-19:
Looks great, good work @aerodolphin !

I worked on this project most of the weekend but this Issue slipped through the net I'm afraid. I'll try to make some progress with it this week.

aerodolphin, 2020-11-02:
@flyte did you made some progress?

BenjiU, 2020-11-04:
Hey, I've created a draft, but i can't test it here. It's on feature branch "feature/MCP23017". may you get it and test it, please? It is based on https://github.com/owainm713/MCP230XX-Python-Module

config should be something like:
```
gpio_modules:
  - name: mcp23017_chip
    module: mcp23017
    i2c_bus_num: 1
    chip_addr: 0x20 //depends on your address...

digital_outputs:
  - name: out2
    module: mcp23017_chip
    pin: 2  # This device is connected to pin 2 of the mcp23017 IO expander
    on_payload: "ON"
    off_payload: "OFF"
```

I haven't added interrupts yet. Do you need them?

Please try it and report the errors here :-)

aerodolphin, 2020-11-04:
I can test it. But I don't know how to install that branch. Can you explain?

Interrupts are good to have, but we can keep this feature for future development.

aerodolphin, 2020-11-05:
I made these steps:

```
cd /home/pi
. ve/bin/activate
git clone https://github.com/flyte/pi-mqtt-gpio.git
cd pi-mqtt-gpio
git checkout feature/MCP23017
```

Inserted your lines in my config file (pi-mqtt-gpio.yml) that is located inside the user pi directory:

```
gpio_modules:
  - name: mcp23017_chip
    module: mcp23017
    i2c_bus_num: 1
    chip_addr: 0x20 //depends on your address...

digital_outputs:
  - name: porta
    module: mcp23017_chip
    pin: 2  # This device is connected to pin 2 of the mcp23017 IO expander
    on_payload: "ON"
    off_payload: "OFF"
```

Run it with this line:

`python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml`

And got this errors:

```
mportError: No module named mcp23017
2020-11-04 22:59:37,096 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 887, in <module>
    main(args)
  File "/home/pi/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 759, in main
    GPIO_MODULES[gpio_config["name"]] = configure_gpio_module(gpio_config)
  File "/home/pi/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 427, in configure_gpio_module
    gpio_module = import_module("pi_mqtt_gpio.modules.%s" % gpio_config["module"])
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named mcp23017
2020-11-04 22:59:40,031 mqtt_gpio (INFO): Startup
```

I also tried to install the mcp23017 module. Don't know if it's needed:

`(ve) pi@raspberrypi:~ $ pip install MCP23017`

I got these errors:
```
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement MCP23017 (from versions: none)
ERROR: No matching distribution found for MCP23017
WARNING: You are using pip version 20.2.3; however, version 20.2.4 is available.
You should consider upgrading via the '/home/pi/ve/bin/python -m pip install --upgrade pip' command.
```

BenjiU, 2020-11-05:
First, please make a `git pull`. I fixed something. But that does not concern your problem.

Are you in the correct directory? May you look, if there is a `pi_mqtt_gpio/modules/mcp23017.py` file?

In the config I removed the // in the chip_addr line, because yaml it doesn't like it.
```
gpio_modules:
  - name: mcp23017_chip
    module: mcp23017
    i2c_bus_num: 1
    chip_addr: 0x20
```

When i start it, i get this output: I think it works (maybe we have to find the correct address), i get the IOError, because I don't have any i2c device.

```
pi@testpi:~/pi-mqtt-gpio $ python -m pi_mqtt_gpio.server config.mcp23017.yml
2020-11-05 10:47:56,552 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1267, in <module>
    main(args)
  File "/home/pi/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1112, in main
    initialise_digital_output(out_conf, GPIO_MODULES[out_conf["module"]])
  File "/home/pi/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 677, in initialise_digital_output
    gpio.setup_pin(out_conf["pin"], PinDirection.OUTPUT, None, out_conf)
  File "pi_mqtt_gpio/modules/mcp23017.py", line 36, in setup_pin
    self.io.set_mode(pin, "output")
  File "pi_mqtt_gpio/modules/mcp23017.py", line 202, in set_mode
    regValue = self.single_access_read(reg)
  File "pi_mqtt_gpio/modules/mcp23017.py", line 100, in single_access_read
    dataTransfer = self.bus.read_byte_data(self.i2cAddress, reg)
  File "/home/pi/.local/lib/python2.7/site-packages/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
  File "/home/pi/.local/lib/python2.7/site-packages/smbus/smbus.py", line 132, in read_byte_data
    raise IOError(ffi.errno)
IOError: 121
Exception IOError: IOError(121,) in <bound method MCP230XX.__del__ of <pi_mqtt_gpio.modules.mcp23017.MCP230XX instance at 0x76220f30>> ignored
```

aerodolphin, 2020-11-05:
Sorry this basic question, but how I can make the git pull? Can you write the complete command line?

BenjiU, 2020-11-05:
Of course: that would be `git pull` in the pi-mqtt-gpio directory.

aerodolphin, 2020-11-05:
I made the `git pull`. Thank you.

Got the same error.:
`ImportError: No module named mcp23017`

The location of the mcp23017.py is:
`/home/pi/pi-mqtt-gpio/pi_mqtt_gpio/modules/mcp23017.py`

Where could be the problem?

BenjiU, 2020-11-05:
Hmm, may you try `make` in pi-mqtt-gpio folder? It makes a schema for me, but fails with black....
Then try it again, please.

BenjiU, 2020-11-05:
Or something is broken with the virtual env? maybe you try it again, but without the first step (https://github.com/flyte/pi-mqtt-gpio/issues/114#issuecomment-652601513) Just clone the repo in another directory...

aerodolphin, 2020-11-05:
With make in `pi-mqtt-gpio` folder I got:

```
python setup.py insert_schema
DEPRECATION: Python 2.7 reached the end of its life on January 1st,
2020. Please upgrade your Python as Python 2.7 is no longer
maintained. pip 21.0 will drop support for Python 2.7 in January 2021.
More details about Python 2 support in pip can be found at
https://pip.pypa.io/en/latest/development/release-process/#python-2-support
pip 21.0 will remove support for this functionality.
running insert_schema
make black
make[1]: Entering directory '/home/pi/pi-mqtt-gpio'
black -l 90 pi_mqtt_gpio
make[1]: black: Command not found
make[1]: *** [Makefile:6: black] Error 127
make[1]: Leaving directory '/home/pi/pi-mqtt-gpio'
make: *** [Makefile:3: schema] Error 2
```

aerodolphin, 2020-11-05:
I will try without the virtual environment. What is the advantage of using the ve?

aerodolphin, 2020-11-05:
Outside the virtual environment I got the same error.

BenjiU, 2020-11-05:
May you copy&paste your whole terminal? Maybe i find something, tomorrow :-)

aerodolphin, 2020-11-05:
I can send it to your e-mail. How can you send me your email?

aerodolphin, 2020-11-05:
I copied the config file pi-mqtt-gpio.yml to pi-mqtt-gpio directory and I got this:

I am using user aerodolphin instead of pi.

```
(ve) aerodolphin@raspberrypi:~/pi-mqtt-gpio $ python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
2020-11-05 21:48:05,716 mqtt_gpio (INFO): Startup
pi_mqtt_gpio/modules/raspberrypi.py:46: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self.io.setup(pin, direction, pull_up_down=pullup, initial=initial)
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/aerodolphin/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1267, in <module>
    main(args)
  File "/home/aerodolphin/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1112, in main
    initialise_digital_output(out_conf, GPIO_MODULES[out_conf["module"]])
KeyError: 'mcp23017'
```

aerodolphin, 2020-11-05:
Outside pi-mqtt-gpio directory:

```
aerodolphin@raspberrypi:~ $ . ve/bin/activate
(ve) aerodolphin@raspberrypi:~ $ python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
2020-11-05 21:53:58,408 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 887, in <module>
    main(args)
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 759, in main
    GPIO_MODULES[gpio_config["name"]] = configure_gpio_module(gpio_config)
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 427, in configure_gpio_module
    gpio_module = import_module("pi_mqtt_gpio.modules.%s" % gpio_config["module"])
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named mcp23017
(ve) aerodolphin@raspberrypi:~ $ 
```
Inside pi-mqtt-gpio directory:

```
aerodolphin@raspberrypi:~ $ . ve/bin/activate
(ve) aerodolphin@raspberrypi:~ $ cd pi-mqtt-gpio
(ve) aerodolphin@raspberrypi:~/pi-mqtt-gpio $ python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
2020-11-05 21:56:28,018 mqtt_gpio (INFO): Startup
pi_mqtt_gpio/modules/raspberrypi.py:46: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self.io.setup(pin, direction, pull_up_down=pullup, initial=initial)
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/aerodolphin/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1267, in <module>
    main(args)
  File "/home/aerodolphin/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1112, in main
    initialise_digital_output(out_conf, GPIO_MODULES[out_conf["module"]])
KeyError: 'mcp23017'
(ve) aerodolphin@raspberrypi:~/pi-mqtt-gpio $ 
```

Log file:

```
2020-11-05 21:57:32,796 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 887, in <module>
    main(args)
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 759, in main
    GPIO_MODULES[gpio_config["name"]] = configure_gpio_module(gpio_config)
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 427, in configure_gpio_module
    gpio_module = import_module("pi_mqtt_gpio.modules.%s" % gpio_config["module"])
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named mcp23017
2020-11-05 21:57:35,708 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 887, in <module>
    main(args)
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 759, in main
    GPIO_MODULES[gpio_config["name"]] = configure_gpio_module(gpio_config)
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 427, in configure_gpio_module
    gpio_module = import_module("pi_mqtt_gpio.modules.%s" % gpio_config["module"])
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named mcp23017
```

aerodolphin, 2020-11-05:
Are you using Python 2 or 3.5?

aerodolphin, 2020-11-06:
Trying two things with MCP23017 IC disconnected.

1. Inside ve directory:
```
(ve) aerodolphin@raspberrypi:~ $ cd ve
(ve) aerodolphin@raspberrypi:~/ve $ python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
2020-11-05 23:28:54,717 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 887, in <module>
    main(args)
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 759, in main
    GPIO_MODULES[gpio_config["name"]] = configure_gpio_module(gpio_config)
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 427, in configure_gpio_module
    gpio_module = import_module("pi_mqtt_gpio.modules.%s" % gpio_config["module"])
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named mcp23017
```
1. Inside ve/pi-mqtt-gpio directory:
```
(ve) aerodolphin@raspberrypi:~/ve $ cd pi-mqtt-gpio
(ve) aerodolphin@raspberrypi:~/ve/pi-mqtt-gpio $ python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
2020-11-05 23:29:28,052 mqtt_gpio (INFO): Startup
pi_mqtt_gpio/modules/raspberrypi.py:46: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self.io.setup(pin, direction, pull_up_down=pullup, initial=initial)
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/aerodolphin/ve/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1267, in <module>
    main(args)
  File "/home/aerodolphin/ve/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1112, in main
    initialise_digital_output(out_conf, GPIO_MODULES[out_conf["module"]])
KeyError: 'mcp23017'
Exception IOError: IOError(121,) in <bound method MCP230XX.__del__ of <pi_mqtt_gpio.modules.mcp23017.MCP230XX instance at 0x75e3aad0>> ignored
(ve) aerodolphin@raspberrypi:~/ve/pi-mqtt-gpio $ 
```

And now inside ve/pi-mqtt-gpio directory with MCP23017 IC connected:
```
(ve) aerodolphin@raspberrypi:~/ve/pi-mqtt-gpio $ python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
2020-11-05 23:36:37,412 mqtt_gpio (INFO): Startup
pi_mqtt_gpio/modules/raspberrypi.py:46: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self.io.setup(pin, direction, pull_up_down=pullup, initial=initial)
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/aerodolphin/ve/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1267, in <module>
    main(args)
  File "/home/aerodolphin/ve/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1112, in main
    initialise_digital_output(out_conf, GPIO_MODULES[out_conf["module"]])
KeyError: 'mcp23017'
(ve) aerodolphin@raspberrypi:~/ve/pi-mqtt-gpio $ 
```

Log file on this last case:
```
2020-11-05 23:40:43,625 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 887, in <module>
    main(args)
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 759, in main
    GPIO_MODULES[gpio_config["name"]] = configure_gpio_module(gpio_config)
  File "/home/aerodolphin/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 427, in configure_gpio_module
    gpio_module = import_module("pi_mqtt_gpio.modules.%s" % gpio_config["module"])
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named mcp23017
```

BenjiU, 2020-11-06:
Okay.
1) Virtuel Environment just helps to keep the "local" python installation clean. When python/pip installs some packages, they are stored in the system. Especially on development and trying different modules, this may "kill" or "floud" your system. Maybe sometimes packages are incompatible.
2) when you run `python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml` outside the pi-mqtt-gpio folder, you got via git and points to branch `feature/MCP23017`, python will use your installed (old) pi_mqtt_gpio version, that does not know mcp23017, so you should get the `ImportError: No module named mcp23017` error. This is okay.
3) when you run `python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml` insidethe pi-mqtt-gpio folder, the "new" module mcp23017 is know. On initializing the outputs, you get a key error:
````
initialise_digital_output(out_conf, GPIO_MODULES[out_conf["module"]])
KeyError: 'mcp23017'
````
That's what we need to investigate: May you post your pi-mqtt-gpio.yml config file? Copy&Paste please, just remove the mqtt info, if there is a password/username. Please don't change any of the other things, maybe there is an error or typo.

aerodolphin, 2020-11-06:
My pi-mqtt-gpio.yml file:

```
mqtt:
  host: 192.168.0.180
  port: 1883
  user: "User"
  password: "Password"
  topic_prefix: home/kitchen

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

  - name: mcp23017_chip
    module: mcp23017
    i2c_bus_num: 1
    chip_addr: 0x20 #depends on your address...

digital_outputs:
  - name: lights
    module: raspberrypi
    pin: 21
    on_payload: "ON"
    off_payload: "OFF"
    initial: low  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    retain: yes # This option value controls if the message is retained. Default is no.

  - name: fan
    module: raspberrypi
    pin: 22
    inverted: yes  # This pin may control an open-collector output which is "on" when the output is "low".
    on_payload: "ON"
    off_payload: "OFF"

  - name: porta
    module: mcp23017
    pin: 2  # This device is connected to pin 2 of the mcp23017 IO expander
    on_payload: "ON"
    off_payload: "OFF"

digital_inputs:
  - name: doorbell
    module: raspberrypi
    pin: 18
    on_payload: "ON"
    off_payload: "OFF"
    pullup: yes
    pulldown: no
    bouncetime: 200
    retain: yes
```

BenjiU, 2020-11-06:
By the outputs, the module must be the module_name from the module section:
```
  - name: porta
    module: mcp23017_chip
    pin: 2  # This device is connected to pin 2 of the mcp23017 IO expander
    on_payload: "ON"
    off_payload: "OFF"
```

aerodolphin, 2020-11-06:
Hi, stupid error. Sorry about that. I will try now after the work when I get back to home. Thanks.

BenjiU, 2020-11-06:
No problem :-) Good luck!

aerodolphin, 2020-11-06:
I confirm that is working. I was able to control an output with Home Assistant. Next I will test some sensors.

aerodolphin, 2020-11-09:
BenjiU, Interrupts are simple to implement?

BenjiU, 2020-11-09:
Hmm, will be some hours, i guess. Are you able to test it?

aerodolphin, 2020-11-09:
It's up to you. I think will be very useful to use this module. And after that I plan to share and distribute the boards. 

I can test it. I'm right now making some more tests to check if it's stable. In terms of hardware and software.

aerodolphin, 2020-11-12:
I made some tests with 32 inputs ans outputs at the same time and it's working well. Only getting some delays. Sometime it is more than 20 seconds. The interrupts will resolve that issue.

BenjiU, 2020-11-12:
Maybe i'll have a look at the interrupts this weekend.

aerodolphin, 2020-11-12:
Good to now. Thanks ;)

jjarokergc, 2020-11-15:
@BenjiU: Thank you for developing this module.  

I assembled a prototype board to test it using three MCP23017 chips wired into 6 outputs and 6 inputs.  The testing used an RPI4B 8GB running Ubuntu 20.04  I also confirm that these 12 I/Os can be controlled by homeassitant over mqtt.

There is an up-to 2 second lag when the six inputs are switched.  This was configured as 2 inputs per chip.  Using 32 inputs, as @aerodolphin mentioned, could have significant polling delay especially with many modules on the I2C bus.  

Could a callback-on-interrupt be used to replace polling?  It would be very desirable for the MCP23017 as there can be up-to 8 of these chips on the I2C bus.

BenjiU, 2020-11-15:
Hi, thanks for your feedback. I haven't had time to do it this weekend, but I thought a lot about it. One problem would be the "configuration" of the interrupt pin: currently the modules don't know each other. So I don't want to add the RaspberryPI into the MCP23017 module for interrupt handler. It should be some generic method.

For the yaml two solutions came to my mind.

1) The interrupt config is in the module section (more logically, I think)
```
mqtt:
  host: 192.168.0.180
...

gpio_modules:
  - name: mcp23017_chip
    module: mcp23017
    i2c_bus_num: 1
    chip_addr: 0x20 #depends on your address...
    interrupt:
          module: raspberrypi
          pin: 2 #pin 2 on raspberry pi module is interrupt pin for mcp23017_chip module
          trigger: rising
          pullup: yes
          pinterrupt: ulldown: no
          bouncetime: 200

digital_inputs:
  - name: doorbell
    module: mcp23017
    pin: 2  # This device is connected to pin 2 of the mcp23017 IO expander
    on_payload: "ON"
    off_payload: "OFF"
    interrupt: falling
```

2) The configuration of the interrupt is in the input section (prevents better double usage of pin 2, maybe)
```
mqtt:
  host: 192.168.0.180
...

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

  - name: mcp23017_chip
    module: mcp23017
    i2c_bus_num: 1
    chip_addr: 0x20 #depends on your address...

digital_inputs:
  - name: doorbell
    module: mcp23017
    pin: 2  # This device is connected to pin 2 of the mcp23017 IO expander
    on_payload: "ON"
    off_payload: "OFF"
    interrupt: falling

  - name: mcp23017_interrupt
    module: raspberrypi
    pin: 2 #pin 2 on raspberry pi module is interrupt pin for mcp23017_chip module
    interrupt_module: mcp23017_chip
    interrupt: rising
```
(I don't know, if the yaml syntax is 100% correct, it's just a dump of my mind)
In the future, we will need a graphical configuration tool: it gets complicated :-D

aerodolphin, 2020-11-28:
Hi @BenjiU,

Any news about the interrupts?

BenjiU, 2020-11-30:
Hi,

not now. We are getting new windows, so I'm busy :-) Also nobody responded to my configure examples above, what would be better...

aerodolphin, 2020-11-30:
Hi,

No problem. I can wait. I'm developing another things at the same time.

Regarding to your question, In my opinion, It's better to have the interrupt configuration in the input section.

BenjiU, 2020-12-17:
I don't think, the polling of inputs, after a input occured, is a good solution. Following situation: two inputs on mcp23017 chips are configured as interrupt inputs. The int pin is connected to rapi input. If a interrupt occurs, you would read the both inputs via i2c, but how do you elevate, which pin generated the interrupt? Maybe it's just a short spike, that's already gone, when you read the inputs.

I think you need a real interrupt service handler, that reads the mcp23017 interrupts register and looks, what interrupts have occured and send a mqtt message for that input.

Sorry guys for slow progress on this topic. But it's always in my mind ;-) Maybe it gets better in the upcoming holidays.

jjarokergc, 2020-12-17:
Here is an example of my home automation/security system that I am integrating that uses four MCP23017 chips.  I hand-soldered these onto a breadboard and interfaced about 16 hardwired sensors.  It is a work-in-progress, but from testing and observation it appears that a "bank-based polling on interrupt" strategy would be optimal.

The MCP23017 has two interrupts (A and B) that signal changes on any of the input pins in Bank A (pins 0-7) or Bank B (pins 8-15).  For example, interrupt A signals a change on any of the pins in Bank A.

Polling all of the pins on Bank A when interrupt A occurs would be sufficient: it would be faster than the current polling approach, and it would be fast enough; and, it would update the state of the changed-pin, even if other pins are polled unnecessarily.  I'm not sure I understand BenjiU's comment, "how do you elevate, which pin generated the interrupt?"  I don't think it is necessary to know which pin's input changed.  That type of intelligence belongs in homeassistant.

The goal is to improve performance (by decreasing delay between an input change and firing a mqtt message).  A "bank-based polling on interrupt" strategy would achieve this by prioritizing polling only onto the groups-of-8 pins where a change occurred.  

[image]

aerodolphin, 2020-12-19:
> Polling all of the pins on Bank A when interrupt A occurs would be sufficient: it would be faster than the current polling approach, and it would be fast enough; and, it would update the state of [...]

I agree with you. We only need to poll the bank where the interrupt occurred. And we don't need to know winch input was.

For sure we will need this part of interrupts working. Because the intend is to used in wired smart homes with a lot of sensors.

Very nice integration @jjarokergc.

aerodolphin, 2020-12-19:
@jjarokergc Can you share a simple schematic of the main components of your hardware?

aerodolphin, 2021-01-21:
@BenjiU any spare time to get back to this project?

flyte, 2021-01-21:
I'm working on an asyncio rewrite of the project, which uses separate coroutines to poll each module. This should speed up polling at least, and I'll implement interrupts soon as well. I don't want to spend a lot of time working on the non-asyncio code, so I'd prefer to take a good look at the interrupts as part of the asyncio work.

I do need people to test the asyncio code, so once I've implemented interrupts on there, I'll update the ticket and hopefully you guys can give it a test.

aerodolphin, 2021-01-22:
Ok @flyte I can test it with MCP23017. ;)

flyte, 2021-01-22:
@aerodolphin 

I've added an untested MCP23017 module to the `feature/asyncio` branch. I know @BenjiU added one before, but I'd prefer to use a library that's maintained by a third party than implement it at a low level in this project.

I have an MCP23017 chip arriving later this evening to test with, but feel free to pull `feature/asyncio` and give it a try. If you run into problems, it might be quicker and better to discuss them on [Discord](https://discord.gg/gWyV9W4). I'll try to keep an eye on it, but @ me on there if you like.

aerodolphin, 2021-01-22:
@flyte  I have basic knowledge about Pyton and linux. For testing purposes I will need to know all the steps for it.

[2 more replies]

closed 2021-02-25 by flyte