flyte/mqtt-io #54: DS18B20 support

patvdleer, 2019-02-20:
Request for support for the DS18B20;

https://github.com/flyte/pi-mqtt-gpio/issues/32#issuecomment-465640303

No, as far as I know, one wire from dht11 and one wire from ds18b20 are incompatible. So you should create a new issue for ds18b20, if you want it to be supported. Maybe you could try a library, that should be supported and mention it there. (e.g. https://github.com/timofurrer/w1thermsensor)

patvdleer, 2019-02-20:
https://github.com/patvdleer/pi-mqtt-gpio/commit/f701a239c9a65804ce4c4358569e4a510f452c68

blademckain, 2019-05-02:
me too need supporto for DS18B20
+1

eXCepT74, 2019-06-20:
Me too and even multi sensor config - multiple on one line as in Tasmota sonoff firmware. I have in plan i2c bus in my house and I have few rooms to check temperature and many open/close / motion  sensors. To extend range I’ll use this:

https://www.sparkfun.com/products/14589

Alycee1993, 2019-07-27:
Is this happening ?

BenjiU, 2020-11-09:
Should be fixed with PR above

closed 2020-11-09 by BenjiU