flyte/mqtt-io #56: How to update (virtualenv install) ?

tomlut, 2019-04-21:
I notice you updated the repository 18 days ago but as a Linux newb I have no idea how to update it on my system.

I followed the  virtualenv installation recommendation.

flyte, 2019-04-21:
You should be able to activate the virtualenv using

`. /path/to/your/virtualenv/bin/activate`

And then use pip to update the software using

`pip install - - upgrade pi-mqtt-gpio`

However, unless you're using a Beaglebone there's no need to update to the latest code, as I was just adding support for that.

closed 2019-04-21 by flyte

tomlut, 2019-04-21:
Awesome. Thanks for the quick response.

tomlut, 2019-04-25:
Another question Flyte. I've just set up a new pi with raspbian lite and am having a bit of trouble. 

`sudo supervisorctl status pi_mqtt_gpio`

Reports that it is running but I'm not getting any mqtt messages published.

I had a look in the log and see a lot of:

```
ImportError: No module named RPi.GPIO
Last will set on 'gpio_bridge_1/status' as 'dead'.
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upg$
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages (0.6.5)
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 557, in <$
    GPIO_MODULES[gpio_config["name"]] = configure_gpio_module(gpio_config)
  File "/home/pi/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 404, in c$
    return gpio_module.GPIO(gpio_config)
  File "/home/pi/ve/local/lib/python2.7/site-packages/pi_mqtt_gpio/modules/raspberrypi.$
    import RPi.GPIO as gpio
```

Which makes no sense to me. How can it not find `RPi.GPIO` if `Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages (0.6.5)`

I tried `sudo apt-get install python-dev python-rpi.gpio` both in the venv and out but was informed that it was already installed.

tomlut, 2019-04-25:
Never mind this fixed it:

```

. mqtt_slave/bin/activate
pip install RPi.GPIO
```