flyte/mqtt-io #57: on-off loop with door sensor

pejotigrek, 2019-05-04:
I am not sure, but it could be related with #48 

I have RPi2 installed in my garage, with latest pi-mqtt-qpio & got [similar to these](https://www.ebay.co.uk/itm/6Pcs-New-MC-38-Wired-Door-Window-Sensor-Magnetic-Switch-Alarm-System-DetectorSP/153424756637) reed switch sensor installed and configured in HomeAssistant.

whenever I open the garage door, sensor instead of keeping itself "open", it gives me on-off-on-off-on... readouts.

    > cat /var/log/pi-mqtt-gpio.log

    Input 'garagedoor' state changed to True
    MQTT client: Sending PUBLISH (d0, q0, r0, m173), 'rpi/input/garagedoor', ... (2 bytes)
    Input 'garagedoor' state changed to False
    MQTT client: Sending PUBLISH (d0, q0, r0, m174), 'rpi/input/garagedoor', ... (3 bytes)
    Input 'garagedoor' state changed to True
    MQTT client: Sending PUBLISH (d0, q0, r0, m175), 'rpi/input/garagedoor', ... (2 bytes)
    Input 'garagedoor' state changed to False
    MQTT client: Sending PUBLISH (d0, q0, r0, m176), 'rpi/input/garagedoor', ... (3 bytes)

I can see the same thing in HomeAssistant's log:

    08:58 Garage door turned off
    08:58 Garage door turned on
    08:58 Garage door turned off
    08:58 Garage door turned on

my pi-mqtt-gpio sensor configuration is:

    gpio_modules:
      - name: raspberrypi
        module: raspberrypi
    
    digital_inputs:
      - name: garagedoor
        module: raspberrypi
        pin: 17
        on_payload: "ON"
        off_payload: "OFF"
        pullup: no
        pulldown: no

I tried with different pullup/pulldown combinations, but they didn't worked at all.

is there any solution at user-side [me, changing some config] or it's more like dev-related thing?

also: it would be **great** to have timestamp next to the log entries just to be able to read issues/problems more precisely :) [or is it my problem, because I didn't set up something?]

flyte, 2019-05-26:
It sounds like the switch is physically bouncing (this happens to most switches to varying degrees). Does it eventually end up on the right value?

If it's bouncing, you can add a capacitor across the sensor terminals, which should help to mitigate it. There's a [good article here](https://www.allaboutcircuits.com/technical-articles/switch-bounce-how-to-deal-with-it/) all about it.

I do think it's important to include this functionality in the software though, so I'll continue to track that on #48 and close this one for now. If you don't think it's an issue with bouncing then please reopen this issue.

Cheers

pejotigrek, 2019-05-26:
hi! thanks for your answer - and I keep my fingers crossed to see this functionality in the software :)

regarding if this is or isn't an issue with bouncing - TBH I guess I have too little knowledge about those parts to be 100% sure. I'm calling it bouncing, because it looks like the closest match - and unfortunately it's not eventually ending on the right value :( the on-and-off relationship lasts as long as my gate is open. sometimes HomeAssistant dashboard doesn't even notice those changes and logbook is the only place I can see that gate was opened [...and closed, and opened, and closed... ;D]

flyte, 2019-05-26:
OK, let's reopen this for the moment then.

I've just added the ability to configure logging in your `config.yml` file. I haven't released a new version yet, but have a look in the readme and try using the software from `develop` branch to test it out.

pejotigrek, 2019-05-28:
thanks! will check it ASAP [I leave for a week+ vacation at friday, so do not have as much time till 12th of June, but will do my best not to keep the test without answers too long].

but - a n00b question: how do I install dev branch? I can see that the installation procedure and package name are identical as master's, am I right?

flyte, 2019-06-06:
You should be able to either clone the repo with git straight to the Pi using something like (I'm on mobile, so from memory):

```
sudo apt-get install git
git clone https://github.com/flyte/pi-mqtt-gpio.git
cd pi-mqtt-gpio
pip install -r requirements.txt
python -m pi_mqtt_gpio.server /path/to/your/config.yml
```

This is assuming you've already activated your virtualenv if you're using one.

pejotigrek, 2019-06-27:
thanks for the explaination!
unfortunately in the meantime SD card used in my _garage-door-sensor-raspberrypi_ died, so I'm building all from scratch with some other stuff installed, and because remote GPIO read component was added in one of the latest HomeAssistant updates - I'm slowly moving towards it. easier to config [I guess - for me] and I can scratch out the middle man [mqtt that is] which also will give me less points of possible errors.

so I'm sorry for bothering you, once again thanks for all help - and I think we can close this issue for now :)

flyte, 2019-06-28:
Fair enough. No problem, and I hope you get something working properly. Cheers

closed 2019-06-28 by flyte