flyte/mqtt-io #62: GPIO Input's Pull up resistor ?

c3reeman, 2019-06-04:
Not a software fault as such just looking for a little advice on switch inputs ( physical switches, doors reed switches etc)

Would a pull up resistor be required say around 4.7k and a small capictor be needed like is commonly used on esp2866 gpis's ?

tomlut, 2019-06-06:
For reed (and other) switches that do not supply a voltage you will need a pull-up resistor. Specifying:

```
    pullup: yes
    pulldown: no
```
in your pin configuration would be sufficient. No need for a physical resistor.

I have sensors located over 30m way hooked up with standard Cat 3, 2-pair telephone wire and have not found the need for a noise reducing resistor-capacitor filter network.

So unless you live in a very electrically noisy environment you wont need it.

flyte, 2019-06-06:
Thanks @tomlut!

@c3reeman as mentioned, I only tend to put capacitors on my inputs when I've noticed some "phantom" input changes that presumably have been caused by things such as high current appliances turning on and off nearby.

There are a couple of useful comments on this here https://www.raspberrypi.org/forums/viewtopic.php?t=117853

c3reeman, 2019-06-09:
Thanks for the advice guys much appreciated, for the cost involved I might use capacitors across all inputs anyway as a precaution, what size have you used @flyte ?

closed 2019-06-11 by c3reeman