flyte/mqtt-io #63: error when using PCF8574

c3reeman, 2019-06-11:
Hi Receiving this error when trying to use expansion boards PCF8574

```
pi@raspberrypi:~ $ python -m pi_mqtt_gpio.server /home/pi/Desktop/config.yml
Last will set on 'pibox/switch/status' as 'dead'.
Collecting pcf8574
  Using cached https://files.pythonhosted.org/packages/5f/ab/c8955e73d866977c221fafc167a2fa6bdab5f07b3569fb2dec1d8a7a8836/pcf8574-0.0.5.tar.gz
Collecting smbus-cffi (from pcf8574)
  Using cached https://files.pythonhosted.org/packages/29/3d/a50bd997979c7554c9c571753d34e976eb88ebf41d3a66accb2468bd3c69/smbus-cffi-0.5.1.tar.gz
    Complete output from command python setup.py egg_info:
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    c/_cffi_backend.c:15:17: fatal error: ffi.h: No such file or directory
     #include <ffi.h>
                     ^
    compilation terminated.
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-1hOYKP/smbus-cffi/setup.py", line 48, in <module>
        'Topic :: System :: Hardware',
      File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 317, in __init__
        self.fetch_build_eggs(attrs['setup_requires'])
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 372, in fetch_build_eggs
        replace_conflicting=True,
      File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 846, in resolve
        dist = best[req.key] = env.best_match(req, ws, installer)
      File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1118, in best_match
        return self.obtain(req, installer)
      File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1130, in obtain
        return installer(requirement)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 440, in fetch_build_egg
        return cmd.easy_install(req)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 699, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 725, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 906, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1145, in build_and_install
        self.run_setup(setup_script, setup_base, args)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1133, in run_setup
        raise DistutilsError("Setup script exited with %s" % (v.args[0],))
    distutils.errors.DistutilsError: Setup script exited with error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-1hOYKP/smbus-cffi/
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 557, in <module>
    GPIO_MODULES[gpio_config["name"]] = configure_gpio_module(gpio_config)
  File "/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 403, in configure_gpio_module
    install_missing_requirements(gpio_module)
  File "/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 228, in install_missing_requirements
    % (module, pkgs_required, err)
__main__.CannotInstallModuleRequirements: Unable to install packages for module <module 'pi_mqtt_gpio.modules.pcf8574' from '/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/modules/pcf8574.pyc'> (['pcf8574']): Command '['/usr/bin/env', 'pip', 'install', 'pcf8574']' returned non-zero exit status 1
pi@raspberrypi:~ $ 
```
My configuration is the same as the one in the example, just using the Pi alone as a module works no problem

flyte, 2019-06-12:
I believe you need `sudo apt-get install libffi-dev`.

c3reeman, 2019-06-12:
Perfect all working now x5 PCF8574 and 48 relays, Out if interest have you had much experience with relay boards from china ? not sure if i just got a bad batch but i had quite a few faulty ones see...

[https://community.home-assistant.io/t/8ch-relays-boards-from-aliexpress-quality-problems/121460](url)

closed 2019-06-12 by c3reeman