flyte/mqtt-io #64: Support for PN532 NFC module

rdehuyss, 2019-06-12:
Hi,

Love this software. Is it possible to support PN532 NFC module via UART?

Thanks,
Ronald

flyte, 2019-06-14:
Hey, glad to hear it!

What type of functionality would be required from this? Would it just be to read a presented RFID tag and publish that over MQTT?

I'm not sure the architecture at the moment would easily support that, but I'm writing an asynchronous version that could probably handle it.

In other words, it's a no at the moment (unless someone comes up with a PR, but it'd have to involve some non-trivial multithreading/processing stuff), but I'll certainly try to include this type of functionality in the asynchronous version. Please don't expect that to be released very soon though.

rdehuyss, 2019-08-04:
Hi, it indeed would just be to read the presented RFID tag and publish that over MQTT. Management of all the tags would be handled somewhere else (homeassistant).

Glad that you would consider this!

closed 2022-11-04 by BenjiU