flyte/mqtt-io #65: No module named dht22 / Type field unknown

SkySails, 2019-06-13:
Hi! I'm trying to publish values from a dht11 sensor  to MQTT using this library and I have encountered some errors. (Clean raspbian lite image). Toggling witches using the raspberry/GPIO module works just fine, but showing the sensors are not. 

Using the example configuration I get this error:
```yaml
Config did not validate:
sensor_inputs:
- 0:
  - type:
    - unknown field
  1:
  - type:
    - unknown field
```
And if i remove the "type" entry from the configuration I am left with the following:
```python
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 570, in <module>
    SENSOR_MODULES[sensor_config["name"]] = configure_sensor_module(sensor_config)
  File "/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 415, in configure_sensor_module
    sensor_module = import_module("pi_mqtt_gpio.modules.%s" % sensor_config["module"])
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named dht22
```
I looked inside the modules directory in this repo and in the correlating file I found "Adafruit_DHT" to be a requirement. This is no surprise and already have that installed but to no avail. What am I missing?

SkySails, 2019-06-14:
Some progress. I managed to get it to execute by installing using the provided setup.py instead of the recommended pip. I did encounter another error implying that dht11 was not a part of the module:
```python 

Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 600, in <module>
    SENSOR_MODULES[sensor_config["name"]] = configure_sensor_module(sensor_config)
  File "/home/pi/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 425, in configure_sensor_module
    return sensor_module.Sensor(sensor_config)
  File "pi_mqtt_gpio/modules/dht22.py", line 35, in __init__
    self.sensor_type = DHTsensor.dht11
AttributeError: 'module' object has no attribute 'dht11'
```
which was easily fixed by changing line `35` in the dht22.py module:
```python
self.sensor_type = DHTsensor.dht11
```
to
```python
self.sensor_type = DHTsensor.DHT11
```

Unfortunately I'm still having trouble. The log is spamming me with this, probably implying that I have some kind of problem with the sensor. 
```
2019-06-14 12:34:37,892 mqtt_gpio (WARNING): sensor_timer_thread: sensor 'temperature' returned null
2019-06-14 12:35:15,822 mqtt_gpio (WARNING): sensor_timer_thread: sensor 'humidity' returned null
```
However, when trying to get a value from the sensor manually using Adafruit_DHT I do get a response.

flyte, 2019-06-14:
Thanks for your detailed report. I'll have a look at the code issues a bit later but my experience with DHT11 sensors is they they sometimes (frequently actually) return null when reading their values. I believe this is something to do with a lack of real-time access to the pins messing up the timing of some of the communication with the DHT11. 

Are the values reported properly sometimes, or does it always report return null?

SkySails, 2019-06-14:
That sounds like the experience I'm having. When polling the sensor for values using WiringPi, almost 90% of the results are 

> Data not good, skip

When using python and Adafruit_DHT however, I __always__ get a result _providing that the script is run with sudo_. This might be where the problem resides.

SkySails, 2019-06-14:
### Success!
The solution was running as root. The sensor now reports back correctly every single time, even with a 2-second interval. Should I do a PR for the required changes to the module?

flyte, 2019-06-14:
Ah, interesting. I wonder if there's a group to which your user can be added in order to allow for real time access. If you test that, remember that you need to log out and back in again for groups to take effect.

A PR would be great, thanks! I see the lowercase issue now. I was wondering why it worked before, but I think the guy who contributed that module used one of the other sensors so I guess it slipped through both of our nets!

Let's leave this issue open for the moment, so we can work out if there's a less drastic solution than running as root.

SkySails, 2019-06-14:
Roger. In the meantime, if I add the humidity part of the sensor to the configuration i get the same value as for the temperature. I'm fairly certain that this is the result of the module's default `type`.
My current config looks like this:
```yaml
  - name: humidity
    module: dht22
    interval: 10 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
    options:
      type: humidity
```
which differs from the example:
```yaml
sensor_inputs:
  - name: dht22_temperature 
    module: dht22
    interval: 10 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
    type: temperature # Can be temperature or humidity
```
The difference is in how the type is defined. Normally, when running as a non-root user, defining the type according to the documentation results in the error mentioned in the first post. Running the server as root with the same configuration instead results in the following error:
```
No handlers could be found for logger "mqtt_gpio"
```
I'm assuming this error to be a side effect of running as root, and that it's trying to output the same error as before (first  post). All this comes down to one question: How am I _supposed_ to define the type?

SkySails, 2019-06-14:
### Solved
This was also easily resolved. This time by editing the schema in `__init__.py`. I added the following to the `sensor_inputs` schema:
```yaml
type:
  type: string
  required: no
  default: temperature
```
and now it works like a charm!

flyte, 2019-06-14:
This sounds like an issue from when I changed the options/type bits around at the end of #61 .

I'm just having a look through it now to see what I've done wrong.

SkySails, 2019-06-14:
The sensor updates are not happening according to the time interval. Updates keep flowing with varied delays between 2 and 7 seconds. Any idea as to what this might have to with?

flyte, 2019-06-14:
Ahh, I added a mechanism for modules to specify extra config schema but didn't allow unknowns in `sensor_inputs`. I've committed that change to `develop` now - are you in a position to check if that fixes your initial problem (the one you already fixed in your **Solved** comment)?

flyte, 2019-06-14:
> The sensor updates are not happening according to the time interval. Updates keep flowing with varied delays between 2 and 7 seconds. Any idea as to what this might have to with?

I didn't write the sensor polling timing part, and it's a little complex to figure out at a glance. I suggest opening a new issue for that, possibly with some timestamped logging output to work from.

SkySails, 2019-06-14:
> Ahh, I added a mechanism for modules to specify extra config schema but didn't allow unknowns in `sensor_inputs`. I've committed that change to `develop` now - are you in a position to check if that fixes your initial problem (the one you already fixed in your **Solved** comment)?

I can confirm that your change fixes that issue for me.
What logging do you want? The log of sensor updates?

flyte, 2019-06-14:
> I can confirm that your change fixes that issue for me.

Excellent!

> What logging do you want? The log of sensor updates?

Yeah, something to work from when debugging the timing part. I must admit that I haven't got around to hooking up a DHT11 and testing the sensor reading code myself, and until now have been relying on the contributors' own testing, so any evidence that I can work from in the mean time would be useful.

commit 0efbd8de (2019-06-14): Corrected lowercase module reference (#66) Fixes #65
closed 2019-06-14 by flyte