flyte/mqtt-io #67: Sensor polling interval not working

SkySails, 2019-06-14:
The polling interval as set in the configuration is not reflected on the update intervals of the sensors when the server is running. The actual interval is sporadic and unnecessary high. 
```
2019-06-14 14:46:02,422 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 42.0
2019-06-14 14:46:09,354 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
2019-06-14 14:46:12,422 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 42.0
2019-06-14 14:46:19,356 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
2019-06-14 14:46:22,424 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 40.0
2019-06-14 14:46:29,349 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
2019-06-14 14:46:32,417 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 40.0
2019-06-14 14:46:39,350 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
2019-06-14 14:46:42,420 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 40.0
2019-06-14 14:46:49,353 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
2019-06-14 14:46:52,421 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 40.0
2019-06-14 14:46:59,353 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
2019-06-14 14:47:02,423 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 40.0
2019-06-14 14:47:09,349 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
2019-06-14 14:47:14,945 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 40.0
2019-06-14 14:47:24,410 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
2019-06-14 14:47:27,480 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 40.0
2019-06-14 14:47:29,349 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
2019-06-14 14:47:37,482 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 40.0
2019-06-14 14:47:39,349 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
2019-06-14 14:47:44,944 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 40.0
2019-06-14 14:47:56,942 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
2019-06-14 14:48:00,011 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 40.0
2019-06-14 14:48:03,081 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
2019-06-14 14:48:06,152 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 39.0
2019-06-14 14:48:09,348 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 28.0
```

flyte, 2019-06-14:
Thanks for the report.

Could you also provide the relevant sections of your config, if you're setting the interval there?

SkySails, 2019-06-14:
I edited the original post to include the wanted information.

flyte, 2019-06-16:
What I think might be happening is that it takes a non-trivial amount of time to read the sensor, so it skews the timing. Can you try setting it to 30 or 60 seconds and see if it outputs the values more punctually?

BenjiU, 2022-11-04:
should be fixed with the async integration. Please create a new issue, if the problem is still there.

closed 2022-11-04 by BenjiU