flyte/mqtt-io #68: DHT22 Module requires root access

SkySails, 2019-06-14:
The DHT22 module in its current form requires root access to provide any value other than `None`.
This is due to its dependency, `Adafruit_DHT`, which needs root privileges to get access to the GPIO interface.

flyte, 2019-06-14:
Thanks for the new issue. You managed to find three separate issues in #65 so this helps keep track of them all.

There's a PR here that was merged in 2016 which reportedly allows non-root access:
https://github.com/adafruit/Adafruit_Python_DHT/pull/42

flyte, 2019-06-14:
There's also a SO post that might help here https://raspberrypi.stackexchange.com/a/40106

Specifically, this line:

```sudo adduser pi gpio```

SkySails, 2019-06-14:
> There's a PR here that was merged in 2016 which reportedly allows non-root access:
> [adafruit/Adafruit_Python_DHT#42](https://github.com/adafruit/Adafruit_Python_DHT/pull/42)

Neat. I wonder why it isn't working for me tho. I can only get Adafruit_DHT to output values when running scripts as root, might have to do with bad permissions on my side? Despite using a fresh install of raspbian lite. 

> Specifically, this line:
>sudo adduser pi gpio

I actually tried that before, but to no avail unfortunately.

flyte, 2019-06-14:
What permissions does /dev/gpiomem have for you?

SkySails, 2019-06-14:
```
pi@terrarium:~ $ ls -ld /dev/gpiomem
crw-rw---- 1 root gpio 247, 0 Jun 14 10:06 /dev/gpiomem
```

flyte, 2019-06-14:
Might be worth rebooting (to make sure the group has taken), and then trying the example from https://github.com/adafruit/Adafruit_Python_DHT/blob/master/examples/AdafruitDHT.py without `sudo` to rule out any issue with pi-mqtt-gpio.

SkySails, 2019-06-14:
That's a VERY good tip. Should've been the first thing i did. I already use that exact file for testing along with manual tests but a reboot is critical!

SkySails, 2019-06-14:
Unfortunately the same result. `Adafruit_DHT.py` fails to get a reading without sudo.

flyte, 2019-06-15:
That's disappointing. Perhaps it's worth creating an issue on the https://github.com/adafruit/Adafruit_Python_DHT/ repo then, to see if they can shed some light on why it won't work without `sudo`.

SkySails, 2019-06-15:
Actually, when deploying the server using supervisor everything works as expected. Running on commandline still throws the same issue but supervisor works.

flyte, 2019-06-15:
Unless you've explicitly set your supervisor config to run as another user,
it'll default to running as root, I believe.

flyte, 2019-06-16:
I've set up a little test rig with some buttons and a dht22 sensor, and I don't have to run as root to be able to read values from the sensor. I'm using a fairly recent version of Raspbian, and version `1.4.0` of `Adafruit-DHT`. The `pi` user was already in the `gpio` group, and it "just worked" for me, I'm afraid!

SkySails, 2019-06-16:
Alright, must be a problem on my end then. Apologies!

closed 2019-06-16 by SkySails

flyte, 2019-06-16:
No probs. Hope you get it sorted.

julioo94, 2019-07-27:
Hello,

I blame you because I try to set up pi-mqtt-gpio with DHT22

I already have gpio switch set up and it works very well

mes as I add the DHT22 and I run the program here is what I put I install the library adafruit

 Config did not validate:
sensor_inputs:
- 0:
  - type:
    - unknown field
  1:
  - type:
    - unknown field

if one of you to an idea to help me on the way is DHT22 it will be great :)