flyte/mqtt-io #69: Help Config?

ViPeR5000, 2019-07-03:
`digital_outputs:
  - name: Ender-Luzes
    module: raspberrypi
    pin: 25
    on_payload: "ON"
    off_payload: "OFF"
    initial: low  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low$
    retain: yes # This option value controls if the message is retained. Default is no.

digital_inputs:
  - name: Ender-Luzes
    module: raspberrypi
    pin: 25
    on_payload: "ON"
    off_payload: "OFF"
    pullup: yes
    retain: yes
    pulldown: no

`

it's possible to use the same topic command  in my case home/octopi/input/Ender-Luzes
to subscribe to remote update.

ViPeR5000, 2019-07-03:
MQTT client: Sending SUBSCRIBE (d0, m2) [('home/octopi/output/Ender-Luzes/set', 1)]
MQTT client: Sending PUBLISH (d0, q0, r1, m1), 'home/octopi/input/Ender-Luzes', ... (3 bytes)
Subscribed to topic: 'home/octopi/output/Ender-Luzes/set'
MQTT client: Sending SUBSCRIBE (d0, m3) [('home/octopi/output/Ender-Luzes/set_on_ms', 1)]
Subscribed to topic: 'home/octopi/output/Ender-Luzes/set_on_ms'
MQTT client: Sending SUBSCRIBE (d0, m4) [('home/octopi/output/Ender-Luzes/set_off_ms', 1)]
Subscribed to topic: 'home/octopi/output/Ender-Luzes/set_off_ms'

flyte, 2020-01-24:
Support has been added by #80 and will be released in the next version of the software.

closed 2020-01-24 by flyte