flyte/mqtt-io #70: Momentary vs rocker switch?

pasleto, 2019-07-07:
Hi, am i able to  connect rocker switch as digital input, or only momentary push button? 

Thanks for the answer :)

BenjiU, 2019-07-08:
Just to be sure: rocker switch means, that it holds its position/state (ON or OFF), until you press it again to opposite state (OFF or ON)?
Shouldn't be a problem. I have this for my light switches.

flyte, 2019-07-08:
Yes, absolutely. Any switch should be suitable. It just pulls the input pin from high to low, or low to high, depending how it's set up.

reestr, 2019-08-05:
I have this same question, how would someone actually configure this though; on receiving an mqtt message drive a GPIO pin from low -> high (1 second) -> low ?

BenjiU, 2019-08-06:
Hi reestr, I don't get your point in total: are you talking about a output? The original question was about an input.

reestr, 2019-08-06:
> Hi reestr, I don't get your point in total: are you talking about a output? The original question was about an input.

You're right, I missed the part about it being for input, sorry for the noise.  I was talking about output, and the same request seems to have been captured [here](https://github.com/flyte/pi-mqtt-gpio/issues/38).

closed 2020-01-24 by flyte