flyte/mqtt-io #71: Lirc for IR control

Odianosen25, 2019-07-14:
Hello,

Will it be possible to add the ability for your module to control IR devices?

Thanks

flyte, 2020-01-02:
Hi @Odianosen25 I think this is out of scope for this project, but the beauty of using MQTT is that you can run multiple smaller bits of software to accomplish what you require. Perhaps try this one https://github.com/hobbyquaker/lirc2mqtt for this.

closed 2020-01-02 by flyte

Odianosen25, 2020-01-03:
Hello @flyte,

Thanks for the response. 

> but the beauty of using MQTT is that you can run multiple smaller bits of software to accomplish what you require

That's the point, I really don't like such systems, where I need to manage multiple services/software to achieve the different things; it reduces reliability in my view, and too much over head. Unless the services are so different.

For example if your module could support LIRC, then I know running it to achieve other things, I can also use to to control IR devices. Lets say I want to use your module on a Pi in a room, as I also have a TV there running Kodi on the Pi. If Lirc was to be a thing, I know just running your module only, on the same Pi running Kodi, will allow me to do all I want in controlling all devices, while utilising the same Pi.

Yes I could use docker and all, but I am more of a "one stop shop" kind of guy. But if you say its out of scope, no worries and thanks again for responding. Though I had already written a little python one which I have [here](https://github.com/Odianosen25/mqtt2irsend) (I am not a professional programmer so forgive how crude it might look ;)).

Kind regards