flyte/mqtt-io #74: HELP WITH AUTO START

Alycee1993, 2019-08-26:
i need help with getting this to start i  am nooob and not really sure how to get this to auto start i hvae tried a few times and doesn't seem to start any time can any one help me

BenjiU, 2019-08-27:
HI, what have you tried, and what didn't work? The "Serving Suggestions" with supervisor from flyte are pretty simple and work fine for me: https://github.com/flyte/pi-mqtt-gpio#serving-suggestion

Alycee1993, 2019-08-28:
hi there i have installed the virtualenv like it says my problem is dont have a this path and tbh i am unsure it it has installed to  but it dose work so my problem is am unsure of the path i would fix the command line
command = /home/pi/ve/bin/python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml

my config.yaml is in the home/pi folder and as i said the script works if i run the command  just unsure of the path 
a

Alycee1993, 2019-08-28:
hi there i have installed the virtualenv like it says my problem is dont have a this path and tbh i am unsure it it has installed to but it dose work so my problem is am unsure of the path i would fix the command line
command = /home/pi/ve/bin/python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml

my config.yaml is in the home/pi folder and as i said the script works if i run the command just unsure of the path

BenjiU, 2019-08-29:
Then your command should look like this:
command = /home/pi/ve/bin/python -m pi_mqtt_gpio.server config.yaml
Because the directory variable already points to /home/pi/
directory = /home/pi

Do you get any error message?

Alycee1993, 2019-08-29:
i dont have folder called ve that my problem 
it says  virtualenv installed fine and so did the scrip  just not sure where it has installed it to 
 if i run  python -m pi_mqtt_gpio.server config.yaml in terminal  it starts fine and works 
is there a file i could search for to get teh path of where it has installed

BenjiU, 2019-08-30:
Okay, after installing virtualenv, you need to create a virtual environment:
```
cd /home/pi
virtualenv ve
. ve/bin/activate
pip install pi-mqtt-gpio
```
first you go to your home directory `cd /home/pi`, then you create the virtual environment in folder '/home/pi/ve' `virtualenv ve`, when this is done, you 'activate' this virtual environment `. ve/bin/activate` and at least, you install pi-mqttt-gpio in this activated environment. Have you executed these steps? Then you should see the /home/pi/ve folder.

BTW: this is only a suggestion: you don't need the virtual environment, you may work without it, than your command in supervisor may look like this:
command = python -m pi_mqtt_gpio.server config.yaml
directory = /home/pi

Alycee1993, 2019-08-30:
i have tried 
command = python -m pi_mqtt_gpio.server config.yml and i get the error   pi_mqtt_gpio                     FATAL     Exited too quickly (process log may h               ave details)
 if i go in the logs  i get   /usr/bin/python: No module named pi_mqtt_gpio

Bettman66, 2019-09-08:
I use a service.
/etc/systemd/system/piface.service
`[Unit]
Description=PiFace2MQTT
After=syslog.target

[Service]
Type=simple
User=pi
Group=pi
WorkingDirectory=/home/pi
ExecStart=/usr/bin/python3 /usr/local/lib/python3.5/dist-packages/pi_mqtt_gpio/server.py /home/pi/config.yml
SyslogIdentifier=piface
StandardOutput=syslog
StandardError=syslog
Restart=always
RestartSec=3

[Install]
WantedBy=multi-user.target`

`sudo systemctl enable piface
sudo systemctl start piface`

flyte, 2020-01-24:
@Alycee1993 I'm closing this issue from inactivity (on my part more than anything). Have you solved this problem? If not, feel free to comment/reopen this issue.

closed 2020-01-24 by flyte

Alycee1993, 2020-01-24:
yeah got it and thats ok