flyte/mqtt-io #75: pulse transmitter only reporting part of pulses

RezzZ, 2019-11-06:
Hi, I just connected my water pulse transmitter to your mqtt-gpio but unfortunately it's only reporting some of the pulses. My pulse transmitter sends a pulse for every liter used resulting in eg 100 liters per day. mqtt-gpio only reported like 50%.. I first thought it was because of the load of the raspberry pi but after moving most of my container to a different pi I still only receive two pulses when consuming 5 liters of water.
Any thoughts on what might be the cause? Is mqtt-gpio not registering because the pulse is too short for it to notice?

I had a very old RPi1 hooked up with an instance of homeassistant reading the pins and hooking it up to a counter and never missed a pulse there.
I already tried switching it to a different pin but no luck either.

the setup:
```
  - name: elster_pr6/ch1p
    module: raspberrypi
    pin: 4
    on_payload: "OFF"
    off_payload: "ON"
    pullup: yes
    retain: yes
```
I inverted the on off logic as a pulse is LOW.

PS. any reason the examples show a pullup and pulldown? would't a 'pullup: yes' already imply a 'pulldown: no'?

RezzZ, 2019-11-06:
hmm it might be to an issue that recently started with buster:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=255887

flyte, 2020-01-02:
> Is mqtt-gpio not registering because the pulse is too short for it to notice?

Most likely, yes. This software polls each input in a loop, so if the pulse occurs inbetween each of the polls (100ms pause between polls), then it won't get picked up. Adding interrupts is a solution to this, but it's a work in progress that most likely won't get implemented for a while.

Any idea how long the pulses last? It might be enough to reduce or remove the sleep between polls.

> any reason the examples show a pullup and pulldown? would't a 'pullup: yes' already imply a 'pulldown: no'?

Yes, it does imply that. I've only included both for completeness but it doesn't require both to work properly.

BenjiU, 2022-11-04:
I'll close this issue, as it is two years old. Please create a new one, if the bug still exists!