flyte/mqtt-io #76: Status of outputs

dcsar1, 2019-11-22:
Is there a way currently to get the current status of an output pin?  In other mqtt devices I am using you can send 'status' as the payload instead of ON or OFF and it will leave the current state as-is and respond with what that state is.

If this is not available, can it be added?

Thanks!  So far I really like pq-mqtt-gpio and plan to use it for some of my home automation project.

BenjiU, 2019-12-29:
Hi, this is currently not implemented, because MQTT is designed as PUSH-only protocol. The MQTT brooker should store all states and transmit it to new subscribers, if configuered. Maybe you need a different design in your project.

flyte, 2020-01-02:
@dcsar1 Exactly as @BenjiU says, this functionality doesn't currently exist on this software.

One of the ways to get around this might be to always 'retain' any messages you send to the `/set` topic for each output. That way, any software that needs to know the output state can subscribe to the same `/set` topic and get updated whenever it changes.

yozik04, 2020-01-23:
I'd word this request a bit differently. When service starts would be good to know current output states. Otherwise I have to post `set` topics to get current values. That would be a must have for homeassistant mqtt discovery.

yozik04, 2020-01-23:
Service restart sets all output states to LOW. How to get this update on hass side?

flyte, 2021-01-24:
I probably should have said to make sure that you've got `retain: yes` in your digital output pin configuration. That way, whenever a client connects to the MQTT server and subscribes to the output's topic, it will receive whatever the last value the output was set to.

Also, if you've got this software set up with an `initial` value for an output, as well as `publish_initial: yes`, then it will set the output to that state when the software starts, and publish the value to MQTT.

I consider this to be a workable solution to this issue so I'm closing it. Feel free to comment for any reason though.

closed 2021-01-24 by flyte