flyte/mqtt-io #77: Garage door with 2 sensors

iondarie, 2019-12-10:
Hello,

I have a garage door that has 2 sensors, one at each end of the line.

One activates when the door is fully open and one when it is fully closed.

Can the status of these 2 sensors be read and posted under the same topic?

Thank you

BenjiU, 2019-12-21:
Hi,

I think, that this is not possible right now. 
The name of the sensor is used as "topic":
```
digital_inputs:
  - name: button
    module: raspberrypi
    pin: 21
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes
```

and in the code the "name" is currently used as "index" in LAST_STATES dict:
```
   for in_conf in digital_inputs:
        print (in_conf)
        initialise_digital_input(in_conf, GPIO_MODULES[in_conf["module"]])
        LAST_STATES[in_conf["name"]] = None
```
So you can't insert two sensors with the same name/topic.

If you're still working on your garage door, maybe we can help you to find a different solution. Greetings.

iondarie, 2020-01-01:
ok so i managed to fix my problem i found a plugin that uses 2 status topics. 

But now a bigger problem, every time i start the script it opens my garage door... tried every possible setting but no luck...

is there something i am doing wrong?

BenjiU, 2020-01-01:
Very hard to tell without any information. Maybe a command is stuck in the mqtt brooker? Maybe you can use `mosquitto_sub ... -v  -t "#"` to look, what is stored there and pushed, when you subscribe your topic? Or may you post the output of your script / pi_mqtt_gpio here?

Your second point may be solved with
- with the set_on_ms option: https://github.com/flyte/pi-mqtt-gpio#temporary-set (preferred)
- a plugin, that sends a "off" after 1 sec or 
- with the issue #38

iondarie, 2020-01-01:
python -m pi_mqtt_gpio.server porti-config.yml
Last will set on 'gates/status' as 'dead'.
MQTT client: Received CONNACK (1, 0)
Connected to the MQTT broker with protocol v3.1.1.
Subscribed to topic: 'gates/output/b_set/set'
Subscribed to topic: 'gates/output/b_set/set_on_ms'
Subscribed to topic: 'gates/output/b_set/set_off_ms'
MQTT client: Sending PUBLISH (dFalse, q1, r1, m4, 'gates/status', ... (7 bytes)
MQTT client: Received SUBACK
MQTT client: Received SUBACK
Input 'b_closed' state changed to True
MQTT client: Sending PUBLISH (dFalse, q0, r1, m5, 'gates/input/b_closed', ... (5 bytes)
MQTT client: Received SUBACK
MQTT client: Received PUBACK (Mid: 4)
Input 'b_open' state changed to False
MQTT client: Sending PUBLISH (dFalse, q0, r1, m6, 'gates/input/b_open', ... (4 bytes)
Input 'b_closed' state changed to False
MQTT client: Sending PUBLISH (dFalse, q0, r1, m7, 'gates/input/b_closed', ... (4 bytes)

iondarie, 2020-01-02:
okay so i managed to make it work, but i have a problem with my reed sensors that keep jumping on/off... like in issue #48  is there any way to work around this?

Thank you

flyte, 2020-01-02:
Glad to hear this is fixed, and thank you again to @BenjiU for helping out :100: 

I'll close this issue, and the followup issue can continue on #48.

closed 2020-01-02 by flyte