flyte/mqtt-io #78: travis build no longer working due to "PyYAML vs. Python is 3.4.8"

BenjiU, 2020-01-02:
Hi flyte,

I'm working on the interrupts (https://github.com/BenjiU/pi-mqtt-gpio/pull/3), but my travis build for py3.4 crashes (see: https://travis-ci.org/BenjiU/pi-mqtt-gpio/builds/632035332?utm_medium=notification&utm_source=github_status). It also happens, when I build my or "your" develop branch.

```
Collecting pytest

  Downloading https://files.pythonhosted.org/packages/48/dd/5a41343e0251ba8fa2293662fdd80377e537ace5576e96d2ca9b06e9ad6f/pytest-4.6.8-py2.py3-none-any.whl (230kB)

Collecting mock

  Downloading https://files.pythonhosted.org/packages/05/d2/f94e68be6b17f46d2c353564da56e6fb89ef09faeeff3313a046cb810ca9/mock-3.0.5-py2.py3-none-any.whl

Collecting paho-mqtt (from -r requirements.txt (line 1))

  Downloading https://files.pythonhosted.org/packages/59/11/1dd5c70f0f27a88a3a05772cd95f6087ac479fac66d9c7752ee5e16ddbbc/paho-mqtt-1.5.0.tar.gz (99kB)

Collecting PyYAML>=4.2b1 (from -r requirements.txt (line 2))

  Downloading https://files.pythonhosted.org/packages/a0/18/842b84b53383d64bffa8e3aa45eaebd8842131d8439fecc82b6d969cda7b/PyYAML-5.3b1.tar.gz (268kB)

PyYAML requires Python '>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.*' but the running Python is 3.4.8
```

May you reproduce this? Your the expert: how to fix this? :-D
Maybe it's because of travis is using a pyYAML preRelease 5.3b candidate and not the latest stable 5.2? https://pypi.org/project/PyYAML/#history

When I fix the requirements.txt to 
'''
paho-mqtt
PyYAML==5.2
enum34
cerberus
'''
it works again... But it's not that nice, is it?

BenjiU, 2020-01-03:
Change it to `PyYAML>=4.2b1,<=5.2` works for me.

flyte, 2020-01-03:
Cool, thanks. I'll make that change here on `develop`. Hopefully it doesn't conflict with yours.

I'd quite like to upgrade the Python version beyond 3.4, but it's important to support whichever Python version is available on Raspbian by default. I expect it's probably gone a bit higher now, but I also don't want to remove support for existing installs.

closed 2020-01-03 by flyte