flyte/mqtt-io #86: No Module dht22 found

Arsenal10108, 2020-02-21:
Kind of a novice at this so be gentle! 

Seems like the exact same issue the SkySails had but he mentioned he was able to get it installed and working using included setup.py folder instead of the recommended pip install. 
I am not sure how to do that so i was wondering if someone could point me in the right direction/tell me how haha

Issue I am having is that when including the type: in the YAML config, Python sees it as an unknown field.
pi@raspberrypi:/ $ python -m pi_mqtt_gpio.server /gpioconfig/config.yml
Config did not validate:
sensor_inputs:
- 0:
  - type:
    - unknown field
  1:
  - type:
    - unknown field

And if I remove the type: field
Last will set on 'home/status' as 'dead'.
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 570, in <module>
    SENSOR_MODULES[sensor_config["name"]] = configure_sensor_module(sensor_config)
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 415, in configure_sensor_module
    sensor_module = import_module("pi_mqtt_gpio.modules.%s" % sensor_config["module"])
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named dht22

I have installed the adafruit_DHT drivers and rebooted since.

Arsenal10108, 2020-02-21:
Making some progress it looks like the PIP install did not add the dht22.py file into the modules directory. I manually created the file and now the server executes. 
I also had to add the allow_unknowns in the __init__.py as mention in #61 
Looks like that got it atlease executable but it throwing this issue now. 

Exception in thread pi-mqtt-gpio_SensorReader:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pi_mqtt_gpio/server.py", line 508, in sensor_timer_thread
    value = round(sensor.get_value(sensor), sens_conf["digits"])
  File "/usr/local/lib/python3.7/dist-packages/pi_mqtt_gpio/modules/dht22.py", line 50, in get_value
    if config["type"] == "temperature" and temperature is not None:
TypeError: 'Sensor' object is not subscriptable

flyte, 2020-02-21:
Hi @Arsenal10108 

You're absolutely right, and it's because I haven't released a new version since that module was added. Sorry about that.

I'm going to go through the changes since the last version and push a new package to PyPI. I'll update you on here once it's done.

Sorry you ended up spending time trying to debug this!

Arsenal10108, 2020-02-21:
It's all good! I am honestly surprised I was able to get that far! Thanks for getting back to me, I'll keep using my tasmotized D1 mini for the DHT22 updates for now.

flyte, 2020-02-22:
I've just made a new version so try upgrading with `pip install --upgrade "pi-mqtt-gpio>=0.4.1"` and report back if it worked for you.

Arsenal10108, 2020-02-22:
Fantastic! Works slick!! I am going to do some research but I figured I would throw it in here while i'm at it. 
Do you know how I can change the output to Fahrenheit since I'm in the US and junk

Arsenal10108, 2020-02-22:
nix that last one! I was able to handle it right within Home Assistant. So no biggy! Thanks!

closed 2020-02-22 by Arsenal10108

flyte, 2020-02-22:
Cool, glad to hear it's working! :tada: