flyte/mqtt-io #88: Config problem

livioavalle, 2020-03-04:
Hi i'm trying to install on home assistant this module, but after doing the command to make the config  (python3 -m pi_mqtt_gpio.server config.yml) it gave me some error and doesnt work.
```
2020-03-04 14:07:27,769 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/usr/lib/python3.8/site-packages/pi_mqtt_gpio/server.py", line 887, in <module>
    main(args)
  File "/usr/lib/python3.8/site-packages/pi_mqtt_gpio/server.py", line 734, in main
    with open(args.config) as f:
FileNotFoundError: [Errno 2] No such file or directory: 'config.yml'

BenjiU, 2020-03-04:
Is your configfile named config.yml and located in the folder, where you execute python3 -m pi_mqtt_gpio.server **config.yml**?

livioavalle, 2020-03-04:
Oh sorry, i think i made a mistake... i didn't understand that i need to make the config file before the execution of the command... sorry

BenjiU, 2020-03-04:
No problem. Just reopen or create a new issue, if you get stuck somewhere.

closed 2020-03-04 by BenjiU