flyte/mqtt-io #89: digital output adds "u" to the topic

nicoolaro, 2020-03-13:
So this is strange problem, everything is integrated to Home Assistant but, push switch ON/OFF seems to be failing. 

error i get is:

> 03-13 19:39:02,518 mqtt_gpio (INFO): Received message on topic u'pizerocam/output/doorbell_led/set': 'OFF'
2020-03-13 19:39:02,523 mqtt_gpio (WARNING): No output found with name of u'doorbell_led''

but my code seems to be ok:
`mqtt:
  discovery: yes
  # The following is optional (homeassistant is the default)
  discovery_prefix: homeassistant
  # Optional too, but change to whatever you like
  discovery_name: MQTT GPIO

  host: 192.168.1.10
  port: 1883
  user: "****"
  password: "****"
  topic_prefix: pizerocam
  status_topic: status
  status_payload_running: running
  status_payload_stopped: stopped
  status_payload_dead: dead
gpio_modules:
  - name: raspberrypi
    module: raspberrypi
    cleanup: yes

digital_inputs:
  - name: doorbell
    module: raspberrypi
    pin: 3
    on_payload: "ON"
    off_payload: "OFF"
    pullup: yes
    pulldown: no
  - name: doorbell_led
    module: raspberrypi
    pin: 2
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes
#    inverted: yes
#    initial: low
#    inverted: yes
    retain: no
    
digital_outputs:
  - name: doorbell_led
    module: raspberrypi
    pin: 2
    on_payload: "ON"
    off_payload: "OFF"
    initial: high
    retain: yes
digital_outputs:
  - name: doorbell
    module: raspberrypi
    pin: 3
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
    retain: yes`

closed 2020-03-13 by nicoolaro

flyte, 2020-03-14:
Assuming you figured this out, but the u part is a red herring. That's just the way Python 2 represents a Unicode string.

The problem you were having (in case someone else has the problem) was that you were using a digital input instead of a digital output, and of course you cannot have a message on MQTT set the state of an input.

nicoolaro, 2020-03-14:
Yes, thank you, that was a problem in my code