flyte/mqtt-io #90: docker error

nicoolaro, 2020-03-14:
I can't get this working on docker.
i tried 

> sudo docker run -ti --rm -v /home/pi/config.yml:/config.yml --privileged flyte/mqtt-gpio
also

> sudo docker run -ti --rm -v /home/pi/config.yml:/config.yml --privileged flyte/mqtt-gpio

my config is in home/pi folder
```
mqtt:
  discovery: yes
  # The following is optional (homeassistant is the default)
  discovery_prefix: homeassistant
  # Optional too, but change to whatever you like
  discovery_name: MQTT GPIO

  host: 192.168.1.10
  port: 1883
  user: "mqtt"
  password: "P@ssword"
  topic_prefix: pizerocam
  status_topic: status
  status_payload_running: running
  status_payload_stopped: stopped
  status_payload_dead: dead
gpio_modules:
  - name: raspberrypi
    module: raspberrypi
    cleanup: yes

digital_inputs:
  - name: doorbell
    module: raspberrypi
    pin: 3
    on_payload: "ON"
    off_payload: "OFF"
    pullup: yes
    pulldown: no
  - name: doorbell_led
    module: raspberrypi
    pin: 2
    on_payload: "ON"
    off_payload: "OFF"
    pullup: yes
    pulldown: no
```

i get error : standard_init_linux.go:211: exec user process caused "exec format error"
Am i following corect installation steps for docker (docker is installed on my Pi Zero W)

flyte, 2020-03-14:
I think the docker image isn't compatible with the ARM processor on the Pi at the moment (I know, it's a bit silly). It's on my list to build an image on the pi itself but I haven't got around to it yet. Sorry about that.

nicoolaro, 2020-03-14:
Thanks, it would be great if we can have that eventually.
Maybe a silly question, Is there another method to start python script on boot except virtualenv or docker as stated in readme?
I'm not too experienced with linux

flyte, 2020-03-14:
You don't necessarily need to use virtualenv (although it's recommended), but supervisor is a relatively straightforward way.

Having said that, I'll see if I can get a docker image built today and will update this ticket with the results.

flyte, 2020-03-14:
I've created the ARM Docker images now, and updated the readme instructions to represent that. Please let me know how you get on.

flyte, 2020-03-14:
Oh, and also have a look at https://github.com/flyte/pi-mqtt-gpio/issues/55 if you run into any problems.

To be honest though, if you want the simplest way of running it, just use Supervisor as recommended in the readme 'Serving Suggestion'.

nicoolaro, 2020-03-14:
> Oh, and also have a look at #55 if you run into any problems.
> 
> To be honest though, if you want the simplest way of running it, just use Supervisor as recommended in the readme 'Serving Suggestion'.

Gotcha, i'm gonna test it right now. It may fail because i'm using Pi Zero which is Arm6 (from what i've read) but i will test both options and report back.
thank you

nicoolaro, 2020-03-14:
tied on RPI3+
Got error below, i may need to look at the thread you posted above

> dietpi@DietPi:~$ sudo docker run -ti --rm -v /home/dietpi/config.yml:/config.yml --privileged flyte/mqtt-gpio:armv7l
> 2020-03-14 16:17:35,888 mqtt_gpio (INFO): Startup
> Collecting RPi.GPIO
>   Downloading RPi.GPIO-0.7.0.tar.gz (30 kB)
> Building wheels for collected packages: RPi.GPIO
>   Building wheel for RPi.GPIO (setup.py) ... done
>   Created wheel for RPi.GPIO: filename=RPi.GPIO-0.7.0-cp36-cp36m-linux_armv7l.whl size=61787 sha256=38606b3d0acbb52dc181d87d5bb3c5f7945677eaafb31dd561989efbb8632891
>   Stored in directory: /home/mqttgpio/.cache/pip/wheels/84/c0/d6/73a6e9f98b9d29edd7c6e981dce47852c3321f34518c80dce1
> Successfully built RPi.GPIO
> Installing collected packages: RPi.GPIO
> Successfully installed RPi.GPIO-0.7.0
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
>     "__main__", mod_spec)
>   File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
>     exec(code, run_globals)
>   File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 901, in <module>
>     main(args)
>   File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 798, in main
>     initialise_digital_input(in_conf, GPIO_MODULES[in_conf["module"]])
>   File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 492, in initialise_digital_input
>     gpio.setup_pin(pin, PinDirection.INPUT, pud, in_conf)
>   File "/home/mqttgpio/pi_mqtt_gpio/modules/raspberrypi.py", line 47, in setup_pin
>     self.io.setup(pin, direction, pull_up_down=pullup, initial=initial)
> RuntimeError: No access to /dev/mem.  Try running as root!

flyte, 2020-03-15:
Ah yeah, that's the same issue as in #55. See if the comments in there fix it for you.

nicoolaro, 2020-03-15:
So i tried supervisor option too, and i just cant make it work. I tried on Pi zero and Pi 3+ too. It seems that i'm definitely doing something wrong.
Last ask , is there a way to run 

> python -m pi_mqtt_gpio.server config.yml

as a system service. I do have another python script running great, but this seems to be a "module", if there's an easy way to write something like:

```
[Unit]
Description=Pi Gpio Mqtt service
After=multi-user.target

[Service]
User=root
Type=idle
ExecStart=/usr/bin/python /home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py /home/pi/config.yml

[Install]
WantedBy=multi-user.target
```

thanks for your help!

flyte, 2020-03-15:
> python -m pi_mqtt_gpio.server config.yml

Just use that line in `ExecStart` I assume. 

> i just cant make it work

If you give specifics I could help. It shouldn't be any harder than using an init script, and has some benefits over it.

nicoolaro, 2020-03-15:
> > python -m pi_mqtt_gpio.server config.yml
> 
> Just use that line in `ExecStart` I assume.
> 
> > i just cant make it work
> 
> If you give specifics I could help. It shouldn't be any harder than using an init script, and has some benefits over it.

followed instructions to the letter.
Supervisor status seems ok:
```
pi@rpi-zero-cam:/etc/supervisor/conf.d $ sudo supervisorctl update
pi_mqtt_gpio: stopped
pi_mqtt_gpio: updated process group
pi@rpi-zero-cam:/etc/supervisor/conf.d $ sudo supervisorctl status
pi_mqtt_gpio                     RUNNING   pid 4915, uptime 0:00:28
```
But no MQTT communicaiton and CPU is stuck at 100% with this:
[image]

Standalone script works

flyte, 2020-03-15:
How long have you left it at 100%? I suspect its just compiling some stuff as part of the RPi.GPIO installation.

nicoolaro, 2020-03-15:
around 20 minutes now

flyte, 2020-03-15:
What's the output in the log file? Should be `/var/log/pi-mqtt-gpio.log` if you're using the config from the readme.

nicoolaro, 2020-03-15:
> 2020-03-15 14:48:04,236 mqtt_gpio (INFO): Startup
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages (0.7.0)
> Traceback (most recent call last):
>   File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
>     "__main__", fname, loader, pkg_name)
>   File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
>     exec code in run_globals
>   File "/home/pi/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 887, in <module>
>     main(args)
>   File "/home/pi/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 759, in main
>     GPIO_MODULES[gpio_config["name"]] = configure_gpio_module(gpio_config)
>   File "/home/pi/ve/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 437, in configure_gpio_module
>     return gpio_module.GPIO(gpio_config)
>   File "/home/pi/ve/lib/python2.7/site-packages/pi_mqtt_gpio/modules/raspberrypi.py", line 19, in __init__
>     import RPi.GPIO as gpio
> ImportError: No module named RPi.GPIO
> 

every 30 seconds

flyte, 2020-03-15:
Looks like you're crossing your global Python install with the one in your virtualenv.

Could you paste your `/etc/supervisor/conf.d/pi-mqtt-gpio.conf` file please?

nicoolaro, 2020-03-15:
same as in readme

> [program:pi_mqtt_gpio]
> command = /home/pi/ve/bin/python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml
> directory = /home/pi
> redirect_stderr = true
> stdout_logfile = /var/log/pi-mqtt-gpio.log

flyte, 2020-03-15:
Looks like you've got RPi.GPIO installed in your global Python environment, but when you run this software in the virtualenv it's not picking it up.

> I do have "standalone" installation too, as from the readme "pip install pi-mqtt-gpio"
if it makes any difference

Yeah, could you remove the `RPi.GPIO` package from your global environment any try again? Easiest way is to make sure you're not in a virtualenv by typing `deactivate`, then do `sudo pip uninstall RPi.GPIO`.

nicoolaro, 2020-03-15:
weird

pi@rpi-zero-cam:~ $ sudo pip uninstall RPi.GPIO
Not uninstalling rpi.gpio at /usr/lib/python2.7/dist-packages, outside environment /usr
Can't uninstall 'RPi.GPIO'. No files were found to uninstall.

flyte, 2020-03-15:
:thinking: 

What is the output of `pip freeze` when you're not in your virtualenv?

nicoolaro, 2020-03-15:
```
appdirs==1.4.3
asn1crypto==0.24.0
Cerberus==1.3.2
configparser==4.0.2
contextlib2==0.6.0.post1
cryptography==2.6.1
distlib==0.3.0
entrypoints==0.3
enum34==1.1.6
filelock==3.0.12
importlib-metadata==1.5.0
importlib-resources==1.3.1
ipaddress==1.0.17
keyring==17.1.1
keyrings.alt==3.1.1
meld3==1.0.2
paho-mqtt==1.5.0
pathlib2==2.3.5
pi-mqtt-gpio==0.5.1
pycrypto==2.6.1
PyGObject==3.30.4
pyxdg==0.25
PyYAML==5.2
RPi.GPIO==0.7.0
scandir==1.10.0
SecretStorage==2.3.1
singledispatch==3.4.0.3
six==1.12.0
supervisor==3.3.5
typing==3.7.4.1
virtualenv==20.0.10
zipp==1.2.0

```

flyte, 2020-03-15:
Try `pip uninstall RPi.GPIO --user`, with or without sudo?

nicoolaro, 2020-03-15:
Usage:
  pip uninstall [options] <package> ...
  pip uninstall [options] -r <requirements file> ...

no such option: --user

hahah, the pains :)

flyte, 2020-03-15:
> hahah, the pains :)

Yeah it's really easy to get into a mess with Python packages when combining `sudo`, global installs and `apt`!

I think it might be easier to use Python 3 for this, since the environments won't clash with any global stuff.

```
deactivate
sudo apt-get install -y python3
cd /home/pi
rm -rf ve
virtualenv --python python3 ve
. ve/bin/activate
pip3 install pi-mqtt-gpio
```

flyte, 2020-03-15:
By the way, there's a really nice short article explaining why we use virtualenvs for Python if you're interested: https://www.dabapps.com/blog/introduction-to-pip-and-virtualenv-python/

nicoolaro, 2020-03-15:
installed python 3.

now when i update supervisor it throws
```

pi@rpi-zero-cam:~ $ sudo supervisorctl update
error: <class 'socket.error'>, [Errno 2] No such file or directory: file: /usr/lib/python2.7/socket.py line: 228
```

do i need to modify anything else

flyte, 2020-03-15:
Oh, did you uninstall anything else from your global Python 2 environment? Now it looks like supervisor has broken :man_facepalming: !

Perhaps try `sudo apt-get install --reinstall supervisor`?

nicoolaro, 2020-03-15:
Same errors with ModuleNotFoundError: No module named 'RPi'
At this point, i gave up and went with system service option.
code for System Service is:

```
[Unit]
Description=Pi Gpio Mqtt service
After=multi-user.target

[Service]
User=pi
Type=idle
ExecStart=python -m pi_mqtt_gpio.server config.yml
WorkingDirectory=/home/pi/
[Install]
WantedBy=multi-user.target
```
It working flawlessly now after system reboot.

@flyte thank so much for your help!

flyte, 2020-03-15:
Hah, OK! Glad you've got it working. It can be a challenge to fix a broken global Python environment.

closed 2020-03-15 by flyte